OpenKinect/libfreenect2 #706: Skeleton tracking

Lofert, 2016-08-12:
Hello everyone, has anyone can get the skeleton data by using libfreenect2 and kinect v2?
Anther question, is the skeleton frame directly obtained from depth sensor in kinect? If so, how can we obtain skeleton data from kinect, just like depth image obtained by libfreenect2?
Thank you!

robozo, 2016-08-12:
Short answer, no.  libfreenect2 is high-quality software, which lets you get depth and RGB data
from the kinect v2 hardware, without using the microsoft driver.

Determining skeleton tracking from depth data is a research problem.  The microsoft driver
does a semi-reasonable job of extracting a skeleton, if you directly face the camera.
But it, for example, can't tell your front from your back.

It's a hard problem!  Maybe someone has collected links to the current state of the art...

chengzijiazu, 2016-08-13:
Here is the solution of skeleton tracking on linux. https://groups.google.com/forum/#!topic/openkinect/cXoEnHDUaZc

Lofert, 2016-08-13:
@robozo Hi，robozo. I wonder if the skeleton data can be got from the kinect v2 hardware just like depth and RGB data.

Lofert, 2016-08-13:
@chengzijiazu Thanks, but i don't use ros.

robozo, 2016-08-14:
Hi @Lofert , no, the kinect just sends depth and RGB data (and audio, i think).
That's amazing enough!  Everything else is software.

Thanks @chengzijiazu, for telling me about ros.  I looked on youtube, looks like the ros
skeleton tracking is no better than the microsoft.  Plenty of room for improvement...

chengzijiazu, 2016-08-29:
lastest paper <Real-time Collection and Analysis of 3-Kinect v2 Skeleton Data in a Single Application>
this paper introduce how to track skeleton with openni2+nite2.

chengzijiazu, 2016-08-29:
Real-time Collection and Analysis of 3-Kinect v2 Skeleton Data in a Single Application

orgicus, 2016-08-31:
I had a very similar question which you can see in this [issue](https://github.com/OpenKinect/libfreenect2/issues/668)

It boiled down to:

1. compiling libfreenect with openni2 support (as per readme)
2. grabbing NiTE2 (I got if from [this repo](https://bitbucket.org/kaorun55/openni-2.2/src/2f54272802bfd24ca32f03327fbabaf85ac4a5c4/NITE%202.2%20%CE%B1/?at=master))
3. placing `libfreenect2/build/lib/libfreenect2-openni2.0.dylib` in `NiTE-MacOSX-x64-2.2/Samples/Bin/OpenNI2/Drivers`

After that I could run samples like `SimpleUserTracker`. Note that I had the issue on OSX, but I imagine this can be adapted to Windows/Linux

HTH

chengzijiazu, 2016-08-31:
thank you! Now I could build the solution.It work well.I just copy the Redist files of Openni2 and NiTE2 to the solution folder.

xlz, 2016-12-05:
Looks like the discussion is over.

closed 2016-12-05 by xlz

fma-1724, 2017-07-19:
Hi,

I'm trying to use kinect v2 with openni and nite with ubuntu
Have any of you know how to do it? or do you know of someone that does?

Thanks

bearpaw, 2019-01-17:
> I had a very similar question which you can see in this [issue](https://github.com/OpenKinect/libfreenect2/issues/668)
>
> It boiled down to:
>
> 1. compiling libfreenect with openni2 support (as per [...]

@fma-1724 

Just put `libfreenect2-openni2.so.0` into the `Samples/Bin/OpenNI2/Drivers` and create a symbolic link `libfreenect2-openni2.so` to it.

boulbasakos, 2022-10-11:
I get black screen when i run ./UserViewer plus when i run ./Protonect with librefreenect2 core gets dumbed but with ./Protonect cpu works fine,Any ideas to help me with this issue?

Thanks