OpenKinect/libfreenect2 #707: Installation issue (step:Build)

Kazl, 2016-08-13:
Everything was going smoothly until building time,
when writing: "cmake.." the following message appears: "CMake Error: The source directory "/libfreenect2" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI."
I can see the CmakeList file in directory and can't figure out why it does that.
Ideas?
/K

Kazl, 2016-08-13:
I reinstalled everything (including homebrew, and ended up even worse :P
though also at the exact same spot, instead of getting Cmake Error, when writing "cmake .." now I only get "no such file or directory" even though I followed the steps thoroughly.

orgicus, 2016-08-30:
Would you mind posting the folder structure you're using ?

I haven't encountered issues on OSX compiling based on the instructions posted in the readme.

closed 2016-10-09 by xlz