OpenKinect/libfreenect2 #708: Get camera space position by color coordinate

kalluwa, 2016-08-17:
I am doing a registration using a marker image:
first detect the marker with RGB buffer, then map the marker coorinates to camera space.
I got a correct result with Kinect SDK2.0 by using == MapColorFrameToCameraSpace ==.
just like this:
[image]

and now i need to implement the same effect with libfreenect2, but being stucked in mapping color coordinate to camera position.

[issue509](https://github.com/OpenKinect/libfreenect2/issues/509) can get the depth with the same coordinate in RGB buffer, but i cannot convert it into camera space.

i don't know to handle this problem, can you give me any suggestions?

kalluwa, 2016-08-17:
i create a map from color to depth and add a 5x3 filter , and i get the correct result:
[image]

closed 2016-08-17 by kalluwa

sairamkolla, 2016-12-13:
Hi @kalluwa  Can you explain what did you mean by 5x3 filter?

floe, 2016-12-13:
A 5x3 upsampling filter is exactly what happens if you use the `bigdepth` parameter for `Registration::apply`.