OpenKinect/libfreenect2 #710: Performance on Jetson TX1

xionglingfeng, 2016-08-22:
What is the expected performance on Jetson TX1? Is there anyone who tried to run libfreenect2 on Jetson TX1 can share the performance data, e.g. FPS?

I am running the latest R24.1, `libfreenect2` cannot detect `TegraJPEG` correctly (seems related to issue #701, but that commit cannot make `TegraJPEG` work on my device, it still not detected). So my configuration is CUDAProcessor (depth) and TurboJPEG (RGB). The performance is very poor. I assume ~5 FPS, and `top` shows ~190% CPU usage.

Anyone has successfully get `TegraJPEG` work? Whats the CPU usage and FPS?

lesterlo, 2016-08-26:
The debug output of the freenect2 said the TegraJPEG RGB packet processor can run in around  180Hz and the CUDA Depth packet processor can run in around 55 - 61Hz.

Abbyls, 2016-09-02:
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [OpenGLDepthPacketProcessor] avg. time: 13.3667ms -> ~74.8129Hz
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 41.353ms -> ~24.1821Hz
and my 4 CPU load are near 90%!
[image]

natejgardner, 2016-10-15:
I'm experiencing the exact same issue. I made my own issue for it [here](https://github.com/OpenKinect/libfreenect2/issues/737). Feel free to merge these issues.

closed 2016-11-02 by xlz