OpenKinect/libfreenect2 #712: Failure to power down device properly

sdenenberg, 2016-08-26:
I installed a second USB 3 controller in order to use 2 kinects at the same time.  Things are going well except I'm running into a problem on the second controller.  When my program crashes (or even when I try to disconnect the kinect properly), the second controller becomes unresponsive and fails to recognize any device at all until I reboot my machine.   This doesn't happen everytime, which leads me to believe this is a timing issue and maybe I'm not disconnecting the kinect properly after all.

This is the new USB controller according to lspci:
03:00.0 USB controller: Etron Technology, Inc. EJ188/EJ198 USB 3.0 Host Controller

To shut down the kinect I make calls to libfreenect2::Freenect2Device::stop and libfreenect2::Freenect2Device::close.

These are the failure to disconnect messages that libfreenect2 logs:

[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.18804ms -> ~122.129Hz
[Error] [usb::TransferPool] failed to cancel transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Debug] [DepthPacketStreamParser] not all subsequences received 255
[Error] [protocol::UsbControl] failed to set ir interface state! LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected). Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] stopping...
[Error] [protocol::UsbControl] failed to set ir interface state! LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected). Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Error] [protocol::UsbControl] failed to release interface with ControlAndRgbInterfaceId(=0)! LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected). Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed

I'm not sure what other information would help in debugging this issue.  I understand that when my program seg-faults that I may exit improperly which could screw up the controller, but I should be able to disconnect the kinect properly otherwise.

Thanks,

Scott

sdenenberg, 2016-08-26:
As a further test, I tried running Protonect with just the single kinect plugged into the new controller - same issue.  When I try to Ctrl-C exit Protonect, it fails in the exact same way, leaving my USB controller useless.

xlz, 2016-08-27:
There is probably little we can help, but can you post relevant dmesg? I think this is kernel driver issues.

sdenenberg, 2016-08-30:
Sorry for not responding sooner, I was away for the weekend without access to the hardware.  Here is the dmesg after crashing:

[  119.394784] xhci_hcd 0000:03:00.0: xHCI host not responding to stop endpoint command.
[  119.394793] xhci_hcd 0000:03:00.0: Assuming host is dying, halting host.
[  119.394888] xhci_hcd 0000:03:00.0: HC died; cleaning up
[  119.394979] usb 3-4: USB disconnect, device number 2
[  119.395697] usb 4-4: USB disconnect, device number 2
[  119.395703] usb 4-4.1: USB disconnect, device number 3

If its a kernel driver issue, who should I talk to?  Is there anything I can upgrade to try to fix it?  The pci usb controller is supposed to work with Ubuntu 14.04.

Thank you for your help,

Scott

sdenenberg, 2016-09-02:
I've tried a few more things, still no luck.  Not sure what the next steps should be.  Any help would be appreciated.

Thanks,

Scott

xlz, 2016-10-09:
EJ188/EJ198 USB 3.0 Host Controller is not a commonly seen USB controller and it is probably less well tested.

The best thing you can try is the latest kernel http://packages.ubuntu.com/yakkety/linux-image. 4.8 maybe.

xlz, 2016-12-05:
NOTOURBUG.

closed 2016-12-05 by xlz