OpenKinect/libfreenect2 #714: OPENNI2+NITE2+libfreenect2

chengzijiazu, 2016-08-30:
Could anyone give me a example code about the openni2+nite2+libfreenect2.
I am trying to perform the openni2 + nite2 + libfreenect2,but it make error with losting the openni2.dll and nite2.dll.

That is the big problem for me .I could not solve it after I had worked it for several days.

Best regard,
Songbin

chengzijiazu, 2016-08-30:
It base on win8.1+vs2015.

chengzijiazu, 2016-08-30:
When I run the NiViewer.exe it can work well.

orgicus, 2016-08-30:
If you have nite2, can you try to :

1. copy `libfreenect2-openni2.0.dll` from `libfreenect2\build\lib` to `YOUR_NiTE_FOLDER\Samples\Bin\OpenNI2\Drivers\`
2. Run the SimpleUserTracker sample ?

Additionally you can tweak NiTE-MacOSX-x64-2.2\Samples\Bin\OpenNI.ini to enable verbose logging and see what's going on (what dll files are expected to be loaded and from what location):

```
[Log]
; 0 - Verbose; 1 - Info; 2 - Warning; 3 - Error. Default - None
Verbosity=0
LogToConsole=1
LogToFile=1

[Device]
;Override=""

[Drivers]
; Location of the drivers specified by a relative path based on OpenNI's shared library or an absolute path.
; Path separator "/" can be used to be portable for any platforms.
; Default - OpenNI2/Drivers
;Repository=OpenNI2/Drivers
```

I can't test on Windows at the moment, but I have recently had issues using NiTE with libfreenect2 on OSX.

chengzijiazu, 2016-08-31:
Thank you! @orgicus  I copy the Redist files to the build solution folder. And it work well!

closed 2016-10-09 by xlz