OpenKinect/libfreenect2 #715: Connecting 2 and 4 kinects for an interactive show

villeevalo, 2016-08-31:
Overview Description:
Hello. Since i'm not the dev in this project, i will write down a brief abt. our project, and hopefully somebody can give us an overview on what's possible. We're working on a set-up with multiple kinects/projectors for an interactive show, in 2 different set-ups - 1st setup features 2 sensors and the 2nd, 4 sensors. Anybody tried out libfreenect2 in such a set-up where you have to connect more than 2 kinects? Does it actually work out? 
Thanks a lot!

floe, 2016-09-02:
Possible, but aou will need a high-end PC with a powerful GPU, and you will need one single USB3 controller for each Kinect, each one with its own PCI-Express x8 or x16 connection.

villeevalo, 2016-09-02:
Thanks for the info. Our config is: 
Level 0 (2 Kinects):  Intel i7-6900K 3.2 GHz, 64GB RAM 4x16GB DDR4 2133MHz C13, 1 x NVIDIA Quadro M4000 8G DDR5, 2 x PCI Express USB BUS 3.0 x16 
Level 1 (4 Kinects): Intel i7-6900K 3.2 GHz, 128GB RAM 8x16GB DDR4 2133MHz C13, 2 x SLI NVIDIA Quadro M4000 8G DDR5, 4 x PCI Express USB BUS 3.0 x16

Does this look right?

floe, 2016-09-02:
Yes, IMHO that should work. IIRC, this is the USB3 card that was used in the setup here: https://www.amazon.de/gp/product/B003MVJG8Q/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1

villeevalo, 2016-09-02:
We actually got this one: https://www.amazon.de/Delock-PCI-Express-Karte-USB/dp/B0037AOTF2/ref=sr_1_fkmr2_2?ie=UTF8&qid=1472817129&sr=8-2-fkmr2&keywords=adapter+delock+pci+express+la+2x+usb+3.0+89243 
Should be ok i guess. Thanks for the help Florian :)

chengzijiazu, 2016-09-02:
Hi @floe ,
If I am going to connect 4 kinect v2,should I buy 4 USB3 cards(Transcend USB 3.0 Schnittstellenkarte/Adapter PCIe (2 Ports)) or just a USB3 cards that include 4 ports?

floe, 2016-09-02:
Each Kinect v2 needs its own USB3 controller, so you actually need to buy 4 separate cards.

chengzijiazu, 2016-09-02:
@floe  Thank you！

chengzijiazu, 2016-09-02:
I am working a project using 4 kinect v2 for rising the FTP to 60.That is my plan.I think it should ok theoretically,but is it a good ideas? @floe [http://note.youdao.com/noteshare?id=b4927e5bdf77daf7c8ea99afb23248b7](url)

chengzijiazu, 2016-09-02:
http://note.youdao.com/share/?id=b4927e5bdf77daf7c8ea99afb23248b7&type=note#/

jasoncatchpole, 2016-09-08:
I'll chime in here. 
As @floe points out you need a USB3 controller for each Kinect, luckily you can get a single PCIe card that has multiple USB controllers.

The following PCIe x4 card has 4xUSB3 ports where each port has its own controller. I can confirm the following works with multiple Kinects:
https://www.amazon.com/Express-SuperSpeed-Adapter-Dedicated-Channels/dp/B00HJZEA2S/ref=sr_1_2?ie=UTF8&qid=1473310532&sr=8-2&keywords=startech+PCIe+usb3

villeevalo, 2016-09-14:
Hi all. Did anybody tested libfreenect2 with unity? if yes, any success there? thanks again

soulslicer, 2016-10-25:
we did it with 4 seperate mini pcs and transferred the data over the network using ROS

XeMinZa, 2017-03-25:
I want to know how to code. Just add a device in the code? like the example?

zx-li001, 2017-04-03:
@jasoncatchpole you have done the testing work, right?

jasoncatchpole, 2017-04-03:
@zx-li001 I have tested the USB3 card I listed above with multiple Kinects yes. Having this number of Kinects starts running into issues with PCIe bandwidth and the number of PCIe lanes your CPU has.