OpenKinect/libfreenect2 #718: Clarifications about skipped packets, lost packets, and not all subsequences received.

ksze, 2016-09-08:
Overview Description:
I would like to get some clarifications about the meaning of these messages:

- `[Debug] [RgbPacketStreamParser] skipping rgb packet!`
- `[Debug] [DepthPacketStreamParser] skipping depth packet`
- `[Debug] [DepthPacketStreamParser] not all subsequences received 1007`
- `[Info] [DepthPacketStreamParser] 1 packets were lost`

What are their implications in practice?

1. Does every _skipped_ packet mean a lost frame?
2. Does every _lost_ packet mean a lost frame?
3. Does every "not all subsequences received" mean a lost frame?
4. Does any of these messages mean a _corrupted_ frame?
5. Does either the freenect2 library or the Kinect device try to re-fetch/resend packets?
6. Sometimes I see a _negative_ number of depth packets lost (`[Info] [DepthPacketStreamParser] -65536 packets were lost`). How is that possible? What does that mean? Is that just a timing bug that's causing an overflow when counting the packets, which means the previous or next "lost" packet is actually not lost?

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`

        83f88b4 Update author list for 0.2 release

2. `uname -a`

        Linux KNPC0116 4.4.0-36-generic #55~14.04.1-Ubuntu SMP Fri Aug 12 11:49:30 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

3. `lsusb -t`
    ```
    /:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
        |__ Port 2: Dev 10, If 0, Class=Hub, Driver=hub/1p, 5000M
            |__ Port 1: Dev 11, If 0, Class=Vendor Specific Class, Driver=usbfs, 5000M
            |__ Port 1: Dev 11, If 1, Class=Vendor Specific Class, Driver=usbfs, 5000M
            |__ Port 1: Dev 11, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 1: Dev 11, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
    /:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
        |__ Port 4: Dev 2, If 0, Class=Wireless, Driver=btusb, 12M
        |__ Port 4: Dev 2, If 1, Class=Wireless, Driver=btusb, 12M
        |__ Port 9: Dev 13, If 0, Class=Hub, Driver=hub/1p, 480M
    /:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
        |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
    /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
        |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
    ```
4. `lspci -nn`
    ```
    00:00.0 Host bridge [0600]: Intel Corporation 4th Gen Core Processor DRAM Controller [8086:0c00] (rev 06)
    00:02.0 VGA compatible controller [0300]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller [8086:0412] (rev 06)
    00:03.0 Audio device [0403]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller [8086:0c0c] (rev 06)
    00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 04)
    00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
    00:16.3 Serial controller [0700]: Intel Corporation 8 Series/C220 Series Chipset Family KT Controller [8086:8c3d] (rev 04)
    00:19.0 Ethernet controller [0200]: Intel Corporation Ethernet Connection I217-LM [8086:153a] (rev 04)
    00:1a.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 [8086:8c2d] (rev 04)
    00:1b.0 Audio device [0403]: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:8c20] (rev 04)
    00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d4)
    00:1c.2 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 [8086:8c14] (rev d4)
    00:1d.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 [8086:8c26] (rev 04)
    00:1f.0 ISA bridge [0601]: Intel Corporation Q85 Express LPC Controller [8086:8c4c] (rev 04)
    00:1f.2 SATA controller [0106]: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] [8086:8c02] (rev 04)
    00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 04)
    06:00.0 Network controller [0280]: Intel Corporation Wireless 7260 [8086:08b1] (rev 6b)
    ```

Steps to Reproduce:

1. Build libfreenect2 with clang 3.8, C++11, OpenCL, and TurboJPEG enabled. No CUDA, no OpenGL.
2. Run `Protonect cl -noviewer`

ksze, 2016-09-08:
Regarding the -65536 packets lost message, I'm starting to suspect it is a simple arithmetic error.
It appears that the sequence of the DepthSubPacketFooter is reset to zero for every 2^16 = 65536 packets? But processed_packets_ keeps getting incremented at https://github.com/OpenKinect/libfreenect2/blob/1aa795b27e53b955089ba6afb0107085a506f741/src/depth_packet_stream_parser.cpp#L123. My guess is that it should be modulo'ed so it is reset just like the footer's sequence.

ksze, 2016-09-08:
In fact, I suspect that the whole of https://github.com/OpenKinect/libfreenect2/blob/1aa795b27e53b955089ba6afb0107085a506f741/src/depth_packet_stream_parser.cpp#L123-L125
need to be corrected like this:

```c++
if (processed_packets_ < 0)
  processed_packets_ = current_sequence_;
else
  processed_packets_ = (processed_packets_ + 1) % 65536;
```

You have been thinking that `processed_packets_++` would get you from the initial value of -1 to 0, in which case you just sync, the value of `processed_packets_` with `current_sequence_`, but the problem is that `processed_packets_` *can* be legitimately reset to zero at a later point, so you can't really say this is your first use of `processed_packets_` based on its value being 0.

xlz, 2016-09-11:
I think I should say something about the design of these messages so the idea does not get fragmented in all these questions.

We wanted to find out what causes data loss so each time when it decides to drop some data it would print some Debug message to indicate the reason. "Skipping" means the data is actually OK at the stream level but the packet decoder is still working on the previous packet and is not ready to take it. Other Debug messages would indicate various types of errors present in the data. On some platforms the performance is so bad that printing one such message for each lost packet would clog the CPU so an Info message "x packets were lost" is added to summarize the number of lost packets without printing one for each.

Usually you would get a few skipping packets at the start because the packet decoders are still initializing, or the first packet is incomplete because it didn't receive the first byte. Then you may sometimes get a few lost packets in the middle because of CPU/GPU latency spikes.

If you get continuous output of lost packet messages, something is probably wrong with the decoders or USB stack, or the decoding is just so slow that a lot of data is unable to get processed.

Usually these messages are not a worry. But if you care about not losing any data, you would want to examine the sequence number of the frames you get to ensure its continuity. Beware that there can also be additional data loss when the data is passed from the library to the user in FrameListeners because the user code is not ready for the next frame and FrameListener would not wait for the user.

> 1. Does every skipped packet mean one lost frame?

Yes. "Skip" means we receive something but voluntarily drop it.

> 2. Does every lost packet mean one lost frame?

Yes. A "packet" is decoded into a "frame". "Lost" means we detect noncontinuous sequence numbers which indicate lost packets caused by nonspecific reasons. Maybe we didn't receive it, or maybe we have dropped it

 > 3. Does any of these messages cover hardware transmission issues (not just because the PC is too busy/too slow to parse/process incoming packets)?

Yes. The way it detects lost packets is to examine the continuity of sequence number. Actually you can do this yourself by looking at the decoded frames and this would even detect frames lost from threading, which is not covered in those message, but as debug information those messages are left there.

> Does every "not all subsequences received" mean a lost frame?

These Debug messages are really debug messages. They are only meaningful for improving the code around them. So you don't really need to worry about them. But anyway a packet is dropped when all its subsequences are not received.

> Does any of these messages mean a corrupt frame that will get passed down the pipeline?

No. Only complete frames are passed down the pipeline. Incomplete ones are dropped/skipped/lost.

> I see that the depth packet can get passed to the packet processor before an "X packets where lost" image is issued.

"X packets where lost" is a summary detecting missing sequence numbers. It is informational.

> Does libusb, libfreenect2, or the Kinect device try to re-fetch/resend packets?

No. Depth transfers are isochronous. Color transfers should resend, or if unable to, report errors.

> Sometimes I see a negative number of depth packets lost 

It's probably a bug in the informational code.

> I also see that the depth and rgb streams use different transfer pools: one uses IsoTransferPool, the other uses BulkTransferPool

These are USB transfer modes https://en.wikipedia.org/wiki/USB#System_design

I agree there is a bug in the counting code. I also agree there is probably some bad coding style over there. To be honest I don't really know and there hasn't been any documentation about the exact behavior of the sequence number we get from the packet content. So far we are guessing it would increment for each packet received in a session, but we don't know if it will start from zero or would wrap around or would stop incrementing when we are not received frames. So I actually don't know what is the correct behavior. This one so far kind of works for the purpose of debugging so I have left it.

closed 2016-09-11 by xlz