OpenKinect/libfreenect2 #719: Map bounding box from depth image to color image

myquestionpp, 2016-09-14:
Hi everyone, 
    For given bounding box from depth image, I would like to find the corresponding bounding box at the color image. But I don't know what formula I can use to find it. Can someone help me? 

Sorry for (maybe) stupid question. I'm just beginner. Thanks in advance!

xlz, 2016-10-09:
You can use the getPointXYZ() function to get the color coordinate of a 3-D point. And then do this for all four corner points.

https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Registration.html#a41a4d8479b97e4c74587ca431a94b2a3

closed 2016-10-09 by xlz

VJoer, 2019-04-23:
But the return value of this function is the actual coordinate, not the color image border corresponding to the border of the depth image. I am also a beginner.