OpenKinect/libfreenect2 #720: Problems building

vkee, 2016-09-16:
Hi,

I'm running into issues trying to make. I am getting a error: ‘memcpy’ was not declared in this scope
   return (char *) memcpy (__dest, __src, __n) + __n;

I do have have all the dependencies installed, including CUDA.

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2015 NVIDIA Corporation
Built on Tue_Aug_11_14:27:32_CDT_2015
Cuda compilation tools, release 7.5, V7.5.17

Error from Terminal.

Scanning dependencies of target generate_resources_tool
[  2%] Building CXX object CMakeFiles/generate_resources_tool.dir/tools/generate_resources.cpp.o
[  5%] Linking CXX executable bin/generate_resources_tool
[  5%] Built target generate_resources_tool
[  8%] Building NVCC (Device) object CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_depth_packet_processor.cu.o
/usr/include/string.h: In function ‘void* __mempcpy_inline(void*, const void*, size_t)’:
/usr/include/string.h:652:42: error: ‘memcpy’ was not declared in this scope
   return (char *) memcpy (__dest, __src, __n) + __n;
                                          ^
CMake Error at cuda_compile_generated_cuda_depth_packet_processor.cu.o.cmake:266 (message):
  Error generating file
  /home/rosie/libfreenect2/build/CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_depth_packet_processor.cu.o

CMakeFiles/freenect2.dir/build.make:295: recipe for target 'CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_depth_packet_processor.cu.o' failed
make[2]: *** [CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_depth_packet_processor.cu.o] Error 1
CMakeFiles/Makefile2:136: recipe for target 'CMakeFiles/freenect2.dir/all' failed
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

xlz, 2016-09-17:
Apparently something like `-D_FORCE_INLINES` is needed.

https://github.com/BVLC/caffe/issues/4046
https://github.com/tensorflow/tensorflow/issues/1346
https://github.com/opencv/opencv/issues/6500

xlz, 2016-10-09:
This will be fixed by #690.

foong, 2016-10-20:
Having similar problems on my macbook
OSX 10.11.6 Elcapitan
NVIDIA GeForce GT 330M 512 MB

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2016 NVIDIA Corporation
Built on Sun_Sep_18_22:16:08_CDT_2016
Cuda compilation tools, release 8.0, V8.0.46

Terminal command:
Wees-MacBook-Pro:build Foong$ make

Terminal Response:
Scanning dependencies of target generate_resources_tool
[  2%] Building CXX object CMakeFiles/generate_resources_tool.dir/tools/generate_resources.cpp.o
[  5%] Linking CXX executable bin/generate_resources_tool
[  5%] Built target generate_resources_tool
[  7%] Building NVCC (Device) object CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_depth_packet_processor.cu.o
nvcc warning : The 'compute_20', 'sm_20', and 'sm_21' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).
nvcc fatal   : The version ('80000') of the host compiler ('Apple clang') is not supported
CMake Error at cuda_compile_generated_cuda_depth_packet_processor.cu.o.RelWithDebInfo.cmake:207 (message):
  Error generating
  /Users/Foong/libfreenect2/build/CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_depth_packet_processor.cu.o

make[2]: *** [CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_depth_packet_processor.cu.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

xlz, 2016-10-20:
@foong Yours is a different one. Open a new issue.

xlz, 2016-10-31:
gcc later than 4.9 is not supported by CUDA 7.5 anymore. So use CUDA 8 instead.

closed 2016-10-31 by xlz