OpenKinect/libfreenect2 #722: color problem for getting pcl::PointXYZRGB

t-lou, 2016-09-20:
While using RegistrationImpl::getPointXYZRGB function in library, the color would be all white. 

My solution is:
  pcl::PointXYZRGB pt;
  registration->getPointXYZ(&undistorted, r, c, pt.x, pt.y, pt.z);
  pt.rgb = *reinterpret_cast<float*>(registered.data + 4 * (r * 512 + c));

Or change "float* registered_data = (float *)registered->data; rgb = registered_data[512*r+c];"
to "rgb = *reinterpret_cast<float*>((float *)registered->data+512*r+c);" in /src/registration.cpp line 329

Cannot help making patch, hopefully this is helpful for some

r9y9, 2016-09-20:
Do you mean [`getPointXYZRGB`](https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Registration.html#a145c5d0d475a688f4c36ec35b1ac4974) doesn't work for you? I'm getting color with that. My code snippet looks like:

```C++
void getPointCloudXYZRGB(libfreenect2::Registration *registration,
                         libfreenect2::Frame *undistorted,
                         libfreenect2::Frame *registered,
                         pcl::PointCloud<pcl::PointXYZRGB>::Ptr cloud) {
    const int w = undistorted->width;
    const int h = undistorted->height;
    cloud->is_dense = false;
    assert(cloud != nullptr);
    pcl::PointXYZRGB *p0 = &cloud->points[0];
    for (size_t ri = 0; ri < h; ++ri) {
        for (size_t ci = 0; ci < w; ++ci) {
            auto p = p0 + w * ri + ci;
            registration->getPointXYZRGB(undistorted, registered, ri, ci, p->x,
                                         p->y, p->z, p->rgb);
        }
    }
}
```

t-lou, 2016-09-20:
No, it doesn't work for me, just all white. The printed `pcl::PointXYZRGB.rgb` also looks strange.

My first code looked like 
```cpp
pcl::PointCloud<pcl::PointXYZRGB>::Ptr cloud(new pcl::PointCloud<pcl::PointXYZRGB>);
cloud->header.frame_id = "world";
cloud->reserve(512 * 424);
for(int r = 0; r < 424; ++r)
{
    for(int c = 0; c < 512; ++c)
    {
        pcl::PointXYZRGB pt;
        registration->getPointXYZRGB(&undistorted, &registered, r, c, pt.x, pt.y, pt.z, pt.rgb);
        cloud->push_back(pt);
    }
}
```
I guess it depends on the compiler. I installed the library with gcc-5, but work usually with clang. But this way should be safer, and also proposed in libpcl.

xlz, 2016-10-09:
```
change
   float* registered_data = (float *)registered->data;
   rgb = registered_data[512*r+c];
to
   rgb = *reinterpret_cast<float*>((float *)registered->data+512*r+c);
```
I don't get it. `registered_data[512*r+c]` is equivalent to `*(registered_data+512*r+c)`, which is then equivalent to `*((float *)registered->data+512*r+c)`. The only additional thing in your change is the reinterpret_cast which should do nothing because the inside is already `float*`.

I don't see the problem.

t-lou, 2016-10-10:
you are absolutely right. avoiding temporary variable used for once is my personal preference, and i just changed type conversion for pointer to casting. it must also work on many computers, but on my laptop it had failed. maybe the memory has been interpreted in other ways for some compilers. 

i think reinterpret_cast might be the suggested way (http://docs.pointclouds.org/1.7.0/structpcl_1_1_point_x_y_z_r_g_b.html). if someone also meets my problem, he may refer to this issue.

my env: arch linux, compiler for installation g++5(other compilers don't give good response in cmake checks), compiler used in project clang++-3.8

xlz, 2016-10-31:
Cant reproduce.

closed 2016-10-31 by xlz

CocoZombie, 2017-09-21:
If you can not reproduce then it does not mean you can close the issue !