OpenKinect/libfreenect2 #725: Building problem with Ubuntu 16.04

Thanh-Binh, 2016-09-21:
By building, I always have errors like:
../lib/libfreenect2.so.0.2.0: undefined reference to 'XineramaIsActive'

Please make sure that I installed glfw3 and this library is found at /usr/local/lib/libglfw3.a

Can anyone help me?
Best thanks

SirDifferential, 2016-09-22:
In my case this was GLFW3 being built statically and that breaking the linking to the X11 libs. Assuming you have those X11 libs installed, rebuild glfw3 and libfreenect2 with the shared lib flag:

```
cmake .. -DBUILD_SHARED_LIBS=ON
make -j 4
sudo make install

cd ~/libfreenect2
mkdir build && cd build
make -j 4
```

xlz, 2016-09-27:
README asks for `apt-get install libglfw3-dev`. This libglfw3.a is an artifact that is produced by yourself.

closed 2016-09-27 by xlz