OpenKinect/libfreenect2 #726: How to improve reliability of depth/ir stream?

ksze, 2016-09-22:
Overview Description:

I have an application adapted from the code for Protonect.

Basically, when I get a new frame, I memcopy the frame data and enqueue the copy into a cyclic buffer, then I have a separate worker thread that works at its own pace and dequeues frames from the cyclic buffer and encodes them using ffmpeg's libav* libraries.

I have `nice(19)`'ed my worker thread that does the encoding, but it still interferes with the depth stream (I get more dropped depth packets than I would like). If I don't perform any encoding, I get a maximum of about 20 dropped depth packets per hour. If I enable the encoding, I can get up to 40 dropped depth packets per hour.

Is there a way I can improve things? I don't really know how libusb works. Would it help if I increase the `nice` priority in `EventLoop::execute()`? I see that it's calling `libusb_handle_events_timeout_completed()`. For instance, what if I did this in `event_loop.cpp`?

```C++
#include <unistd.h>

/*...*/

void EventLoop::execute()
{
  this_thread::set_name("USB");
  timeval t;
  t.tv_sec = 0;
  t.tv_usec = 100000;

  nice(-10); // increase the nice priority of this thread
  while(!shutdown_)
  {
    libusb_handle_events_timeout_completed(reinterpret_cast<libusb_context *>(usb_context_), &t, 0);
  }
}
```

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`

        83f88b4c09f0b00724ae65785abcd4f3eeb79f93

2. `uname -a`

        Linux KNPC0116 4.4.0-36-generic #55~14.04.1-Ubuntu SMP Fri Aug 12 11:49:30 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

3. `lsusb -t`

        /:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
            |__ Port 2: Dev 10, If 0, Class=Hub, Driver=hub/1p, 5000M
                |__ Port 1: Dev 11, If 0, Class=Vendor Specific Class, Driver=usbfs, 5000M
                |__ Port 1: Dev 11, If 1, Class=Vendor Specific Class, Driver=usbfs, 5000M
                |__ Port 1: Dev 11, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
                |__ Port 1: Dev 11, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
        /:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
            |__ Port 4: Dev 2, If 0, Class=Wireless, Driver=btusb, 12M
            |__ Port 4: Dev 2, If 1, Class=Wireless, Driver=btusb, 12M
            |__ Port 9: Dev 13, If 0, Class=Hub, Driver=hub/1p, 480M
        /:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
            |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
            |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M

4. `lspci -nn`

        00:00.0 Host bridge [0600]: Intel Corporation 4th Gen Core Processor DRAM Controller [8086:0c00] (rev 06)
        00:02.0 VGA compatible controller [0300]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller [8086:0412] (rev 06)
        00:03.0 Audio device [0403]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller [8086:0c0c] (rev 06)
        00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 04)
        00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
        00:16.3 Serial controller [0700]: Intel Corporation 8 Series/C220 Series Chipset Family KT Controller [8086:8c3d] (rev 04)
        00:19.0 Ethernet controller [0200]: Intel Corporation Ethernet Connection I217-LM [8086:153a] (rev 04)
        00:1a.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 [8086:8c2d] (rev 04)
        00:1b.0 Audio device [0403]: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:8c20] (rev 04)
        00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d4)
        00:1c.2 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 [8086:8c14] (rev d4)
        00:1d.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 [8086:8c26] (rev 04)
        00:1f.0 ISA bridge [0601]: Intel Corporation Q85 Express LPC Controller [8086:8c4c] (rev 04)
        00:1f.2 SATA controller [0106]: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] [8086:8c02] (rev 04)
        00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 04)
        06:00.0 Network controller [0280]: Intel Corporation Wireless 7260 [8086:08b1] (rev 6b)

xlz, 2016-09-27:
40 per hour is not bad at all. Your requirement is simply very strict.

I think the mostly likely case is GPU having latency spike. Memory defrag may also cause issues.

I can't pinpoint the exact location that is dropping packets. You would need to add custom timestamping at various levels in libfreenect2 to examine this.

You can try setting realtime priority instead of nice, but this would only get rid of CPU scheduling latency, not USB controller, not GPU.

xlz, 2016-10-09:
If you have hard requirements on getting every last bit of data, there is an alternative route. You can record the raw USB data and perform the decoding offline/non-realtime. libfreenect2 does realtime decoding now and will drop data if decoding is too slow.

closed 2016-10-09 by xlz