OpenKinect/libfreenect2 #727: GLFW linking error on Mac OS X El Capitan

TheInnerParty, 2016-09-24:
Overview Description: libfreenect2 won't link properly during compile on Mac OS X El Capitan due to GLFW problem.

Version, Platform, and Hardware Bug Found:

1.  git log -1 --oneline :

`83f88b4 Update author list for 0.2 release`

2. uname -a:

`Darwin computerName 15.6.0 Darwin Kernel Version 15.6.0: Thu Jun 23 18:25:34 PDT 2016; root:xnu-3248.60.10~1/RELEASE_X86_64 x86_64`

3. lsusb -t:
N/A

4. lspci -nn:
N/A

Steps to Reproduce:

1. Install all the dependencies, specifically, libusb, the latest version of GLFW (3.2.1), and OpenNI
2. Generate makefiles with cmake
3. build with make

Actual Results:

Compiler Outputs this error: 
~~~~
[ 63%] Building CXX object CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o
[ 65%] Linking CXX shared library lib/libfreenect2.dylib
Undefined symbols for architecture x86_64:
  "_CGAcquireDisplayFadeReservation", referenced from:
      _beginFadeReservation in libglfw3.a(cocoa_monitor.m.o)
  "_CGAssociateMouseAndMouseCursorPosition", referenced from:
      __glfwPlatformSetCursorMode in libglfw3.a(cocoa_window.m.o)
  "_CGDisplayBounds", referenced from:
      _acquireMonitor in libglfw3.a(cocoa_window.m.o)
      _transformY in libglfw3.a(cocoa_window.m.o)
      __glfwPlatformGetMonitorPos in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayCopyAllDisplayModes", referenced from:
      __glfwSetVideoModeNS in libglfw3.a(cocoa_monitor.m.o)
      __glfwPlatformGetVideoModes in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayCopyDisplayMode", referenced from:
      __glfwSetVideoModeNS in libglfw3.a(cocoa_monitor.m.o)
      __glfwPlatformGetVideoMode in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayFade", referenced from:
      _beginFadeReservation in libglfw3.a(cocoa_monitor.m.o)
      _endFadeReservation in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayGammaTableCapacity", referenced from:
      __glfwPlatformGetGammaRamp in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayIOServicePort", referenced from:
      _getDisplayName in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayIsAsleep", referenced from:
      __glfwPlatformGetMonitors in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayModeCopyPixelEncoding", referenced from:
      _modeIsGood in libglfw3.a(cocoa_monitor.m.o)
      _vidmodeFromCGDisplayMode in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayModeGetHeight", referenced from:
      _vidmodeFromCGDisplayMode in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayModeGetIOFlags", referenced from:
      _modeIsGood in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayModeGetRefreshRate", referenced from:
      _vidmodeFromCGDisplayMode in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayModeGetWidth", referenced from:
      _vidmodeFromCGDisplayMode in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayModeRelease", referenced from:
      __glfwPlatformGetVideoMode in libglfw3.a(cocoa_monitor.m.o)
      __glfwRestoreVideoModeNS in libglfw3.a(cocoa_monitor.m.o)
  "_CGDisplayMoveCursorToPoint", referenced from:
      __glfwPlatformSetCursorPos in libglfw3.a(cocoa_window.m.o)
[... 156 lines cut ...]
      __glfwPlatformGetWindowPos in libglfw3.a(cocoa_window.m.o)
      -[GLFWContentView mouseMoved:] in libglfw3.a(cocoa_window.m.o)
      -[GLFWContentView viewDidChangeBackingProperties] in libglfw3.a(cocoa_window.m.o)
      -[GLFWContentView updateTrackingAreas] in libglfw3.a(cocoa_window.m.o)
      -[GLFWContentView performDragOperation:] in libglfw3.a(cocoa_window.m.o)
      -[GLFWContentView firstRectForCharacterRange:actualRange:] in libglfw3.a(cocoa_window.m.o)
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libfreenect2.0.2.0.dylib] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2
~~~~

Expected Results:

The Code should build.

Reproducibility:

I tried rebuilding GLFW with the ` -DBUILD_SHARED_LIBS=ON` as per another issue suggestion, but that didn't help at all. It gives the same error every time.

Additional Information:

Thanks for any tips on fixing this!

xlz, 2016-10-09:
There is no instruction for you to build GLFW3 from source on MacOSX. Where did you get libglfw3.a?

Anyway you need to remove libglfw3.a wherever this is.

xlz, 2016-12-05:
I believe this is NOTABUG.

closed 2016-12-05 by xlz