OpenKinect/libfreenect2 #728: error while loading shared library libfreenect2.so.0.2

ashokbugude, 2016-09-27:
I am using ubuntu 16.04. I am trying to run Protonect.cpp from 'Code Blocks'.I am able to build without any errors, but when I run , I get below errors

Error while loading shared library libfreenect2.so.0.2 
[image]

I have added the libfreenect2.so.0.2  file in linkers section of codeblocks as shown below, but still get same error
[image]

Please help resolve the issue

floe, 2016-09-27:
What output do you get when you run `ldd /home/zayyon/Ashok/codeBlocksWS/test/bin/Debug/test` in the console?

ashokbugude, 2016-09-27:
Thanks for your response, Below is the output of ldd command
[image]

I can see 'libfreenect2.so.0.2' is not found although I have added it in 'Linker settings' of codeblocks as shown in previous screenshot. Please help setup in codeBlocks

floe, 2016-09-27:
Sorry, no idea about codeBlocks. But this means that `/usr/local/lib` is not part of your library search path, for whatever reason. As a temporary fix, you can do `export LD_LIBRARY_PATH=/usr/local/lib` in the console and run the binary from there.

ashokbugude, 2016-09-27:
Thanks, working fine from terminal but not from codeblocks

closed 2016-09-27 by ashokbugude