OpenKinect/libfreenect2 #729: What is the exposure meaning in libfreenect2?

Edwinzero, 2016-09-28:
Overview Description:

I find the exposure definition at frame_listener.hpp

  float exposure;         ///< From 0.5 (very bright) to ~60.0 (fully covered)
  float gain;             ///< From 1.0 (bright) to 1.5 (covered)
  float gamma;            ///< From 1.0 (bright) to 6.4 (covered)

I think 0.5 means 2ms exposure time. and 60 means 1/60 ms exposure time.
Could someone explain this range for me?

xlz, 2016-09-28:
0.5 means 0.5ms? 60 means 60ms?

closed 2016-10-07 by xlz

Edwinzero, 2016-10-17:
@xlz , I am trying to use the color exposure and gamma information do some test. I saw the comment says that 0.5 means very bright, and 60 means fully covers. Theoretically, the longer shutter time, the higher exposure time. That is why I think that 0.5 is means 1/0.5 ms = 2ms , and 60 means 1/60 ms.

xlz, 2016-10-17:
The camera has auto exposure. If the environment is very bright, it would lower the exposure time and vice versa.