OpenKinect/libfreenect2 #730: libusb package not found/fails version check during installation (Raspbian)

ibranch, 2016-10-03:
When installing libfreenect 2 on a fresh version of Raspbian, (Linux kernel version 4.44.22-v7+, result of `uname -a` is: Linux raspberrypi 4.4.22-v7+ #912 SMP Mon Sep 26 19:04:59 BST 2016 armv7l GNU/Linux), the `cmake` command fails because cmake cannot find a required package, namely libusb-1.0.20 or above, with the following error:

```-- checking for module 'libusb-1.0>=1.0.20'
--   package 'libusb-1.0>=1.0.20' not found
CMake Error at /usr/share/cmake-3.0/Modules/FindPkgConfig.cmake:341 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.0/Modules/FindPkgConfig.cmake:395 (_pkg_check_modules_internal)
  cmake_modules/FindLibUSB.cmake:21 (PKG_CHECK_MODULES)
  CMakeLists.txt:93 (FIND_PACKAGE)
```
and further down in the output:
```CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LibUSB_LIBRARY
    linked by target "freenect2" in directory /home/pi/libfreenect2
```

Note that this occurs regardless of the flags passed to cmake, and I am running it from `~/libfreenect2/build` as `cmake ..`, where `~/libfreenect2/` is a clone of this project, today.

Running `sudo apt-get install libusb1.0-0-dev` informs me that I am already on the newest version.

Checking libusb.org, I see that they have only released up to version 1.0.9, so I'm not sure why cmake is checking for version 1.0.20, which would seem to not exist.

the result of `dpkg -l | grep libusb` is:
```ii  libusb-0.1-4:armhf                    2:0.1.12-25                               armhf        userspace USB programming library
ii  libusb-1.0-0:armhf                    2:1.0.19-1                                armhf        userspace USB programming library
ii  libusb-1.0-0-dev:armhf                2:1.0.19-1                                armhf        userspace USB programming library development files
ii  libusb-1.0-doc                        2:1.0.19-1                                all          documentation for userspace USB programming
ii  libusbmuxd2:armhf                     1.0.9-1                                   armhf        USB multiplexor daemon for iPhone and iPod Touch devices - library
```

I don't know enough about cmake or the setting and testing of variables in CMake files to debug this on my own. Please help!

xlz, 2016-10-04:
1.0.20 is required. 1.0.19 is not accepted.

ibranch, 2016-10-04:
How would you suggest one install version 1.0.20? As I've said, following the installation instructions results in the above packages being installed and cmake failing. 

Also, why do you reference 1.0.19? I can't find anything higher than 1.0.9 which I found in a tarball at libusb.org

ibranch, 2016-10-04:
Nevermind, I've found a sourceforge version. Please update the Linux installation documentation to point at 

https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.20/libusb-1.0.20.tar.bz2/download

Or some mirror of that file, as the Linux installation instructions result in failure currently. 

I can submit a pull request to the README when I'm at a computer, commenting from my phone and not savvy enough to do it from here. 

Thanks for all your help xlz

xlz, 2016-10-04:
You should install from raspbian repo

http://archive.raspbian.org/raspbian/pool/main/libu/libusb-1.0/libusb-1.0-0-dev_1.0.20-1_armhf.deb
http://archive.raspbian.org/raspbian/pool/main/libu/libusb-1.0/libusb-1.0-0_1.0.20-1_armhf.deb

closed 2016-10-04 by xlz