OpenKinect/libfreenect2 #732: Problems building

vkee, 2016-10-04:
Hi,

I'm running into issues trying to make. I am getting a error: : fatal error: helper_math.h: No such file or directory

[  5%] Built target generate_resources_tool
[  8%] Building NVCC (Device) object CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_depth_packet_processor.cu.o
/home/robbie/libfreenect2/src/cuda_depth_packet_processor.cu:31:25: fatal error: helper_math.h: No such file or directory
compilation terminated.
CMake Error at cuda_compile_generated_cuda_depth_packet_processor.cu.o.cmake:207 (message):
  Error generating
  /home/robbie/libfreenect2/build/CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_depth_packet_processor.cu.o

CMakeFiles/freenect2.dir/build.make:68: recipe for target 'CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_depth_packet_processor.cu.o' failed
make[2]: *** [CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_depth_packet_processor.cu.o] Error 1
CMakeFiles/Makefile2:136: recipe for target 'CMakeFiles/freenect2.dir/all' failed
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

closed 2016-10-05 by vkee

natejgardner, 2016-10-15:
I am having the same issue. How did you solve this? @vkee

vkee, 2016-10-15:
I don't remember exactly, but I think you need to make sure that in the CMake configuration, it's pointing to the correct CUDA library directory.

natejgardner, 2016-10-22:
I solved this by using this fork: https://github.com/geoffviola/libfreenect2

vkee, 2016-11-04:
On a new separate machine, I ran into similar issues and that fork fixed the issue!

For others/future reference, this is what it said:

[  6%] Building NVCC (Device) object CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_depth_packet_processor.cu.o
nvcc warning : The 'compute_20', 'sm_20', and 'sm_21' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).
nvcc warning : The 'compute_20', 'sm_20', and 'sm_21' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).
/usr/lib/gcc/x86_64-linux-gnu/4.8/include/stddef.h(432): error: identifier "nullptr" is undefined

/usr/lib/gcc/x86_64-linux-gnu/4.8/include/stddef.h(432): error: expected a ";"

/usr/include/x86_64-linux-gnu/c++/4.8/bits/c++config.h(190): error: expected a ";"

/usr/include/c++/4.8/exception(63): error: expected a ";"

/usr/include/c++/4.8/exception(68): error: expected a ";"

/usr/include/c++/4.8/exception(76): error: expected a ";"

/usr/include/c++/4.8/exception(83): error: expected a ";"

/usr/include/c++/4.8/exception(93): error: expected a "{"

/usr/include/c++/4.8/bits/exception_ptr.h(64): error: function "std::current_exception" returns incomplete type "std::__exception_ptr::exception_ptr"

/usr/include/c++/4.8/bits/exception_ptr.h(64): error: expected a "{"

/usr/include/c++/4.8/bits/exception_ptr.h(79): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(81): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(82): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(84): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(86): error: declaration is incompatible with previous "std::current_exception"
(64): here

/usr/include/c++/4.8/bits/exception_ptr.h(86): error: use of a local type to declare a function

/usr/include/c++/4.8/bits/exception_ptr.h(86): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(87): error: use of a local type to declare a function

/usr/include/c++/4.8/bits/exception_ptr.h(90): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(92): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(95): error: incomplete type is not allowed

/usr/include/c++/4.8/bits/exception_ptr.h(95): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(116): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(126): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(143): error: use of a local type to declare a function

/usr/include/c++/4.8/bits/exception_ptr.h(144): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(147): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(152): error: use of a local type to declare a function

/usr/include/c++/4.8/bits/exception_ptr.h(153): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(156): error: use of a local type to declare a function

/usr/include/c++/4.8/bits/exception_ptr.h(157): error: expected a ";"

[... 132 lines cut ...]

/usr/include/c++/4.8/cmath(375): error: expected a ";"

/usr/include/c++/4.8/cmath(406): error: inline specifier allowed on function declarations only

/usr/include/c++/4.8/cmath(406): error: variable "std::constexpr" has already been defined

/usr/include/c++/4.8/cmath(406): error: expected a ";"

/usr/include/c++/4.8/cmath(443): error: inline specifier allowed on function declarations only

/usr/include/c++/4.8/cmath(443): error: variable "std::constexpr" has already been defined

/usr/include/c++/4.8/cmath(443): error: expected a ";"

Error limit reached.
100 errors detected in the compilation of "/tmp/tmpxft_00002ad4_00000000-7_cuda_depth_packet_processor.cpp1.ii".
Compilation terminated.
CMake Error at cuda_compile_generated_cuda_depth_packet_processor.cu.o.cmake:264 (message):
  Error generating file
  /home/momap/libfreenect2/build/CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_depth_packet_processor.cu.o

make[2]: *** [CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_depth_packet_processor.cu.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

xiekuncn, 2016-11-24:
@vkee I had same issue as you . and I notice the version has a typo VER**S**ION in at Line 359  CMakelists.txt. 
So I delete that IF block for this issue. and I can pass the compiling. 
```
    -  IF(CUDA_VERSION VERSION_LESS 7.0)
    -   STRING(REGEX REPLACE "-std=c\\+\\+.." "" CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS}")
    - ENDIF()
    + STRING(REGEX REPLACE "-std=c\\+\\+.." "" CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS}")
```

RajaMOeez, 2017-02-20:
hello @xiekuncn my problem is that Im building it vs2013 in windows 8.1 only one error is occur in 
**C:\Users\UserName\Documents\Visual Studio 2013\Projects\libfreenect2-master\libfreenect2-master\src\Cuda_Depth_packet_processor.cu(31): fetal error c1083: cannot open include helper_math.h**  

Is any help for this I'll be very grateful to you 
Best Regards 
MOeez Raja

fidergo-stephane-gourichon, 2019-04-02:
Had same problem here on Ubuntu 16.04.

The fork mentioned simply provides a local copy `helper_math.h`, see https://github.com/geoffviola/libfreenect2/commit/6626278639fda68488110c9c478dcc37be0b3bd9

This feels wrong because if the original did not provide a local copy, there is probably a reason (redistribution restriction, perhaps). Anyway, the mentioned fork lags behind original repository (so far, 65 commits).

Anyway, I fixed like this:

Find where the original file is located:

```bash
$ find /usr/local/ -xdev -iname helper_math.h
/usr/local/cuda-10.1/samples/common/inc/helper_math.h
```

Then instruct freenect2 where to find it. I simply added these words in my CMake command line:

```bash
-D CUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-10.1/ \
```

Problem cleanly fixed, and still using official, up-to-date git repository.

AndiDemon, 2024-12-27:
Had a problem with another one after I fixed the helper_math.h error.

I used the gcc 12.30 and cuda 12

`[  7%] Building CXX object CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o
/usr/bin/c++ -DRESOURCES_INC -Dfreenect2_EXPORTS -I/home/amd/libfreenect2/include -I/home/amd/libfreenect2/include/internal -I/home/amd/libfreenect2/build -I/home/amd/libfreenect2/src/tinythread -I/usr/include/libusb-1.0 -Wall -fno-exceptions -g -fPIC -fvisibility=hidden -fvisibility-inlines-hidden -MD -MT CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o -MF CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o.d -o CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o -c /home/amd/libfreenect2/src/opencl_depth_packet_processor.cpp
In file included from /usr/include/CL/cl.h:20,
                 from /usr/include/CL/opencl.h:24,
                 from /home/amd/libfreenect2/include/internal/CL/cl.hpp:175,
                 from /home/amd/libfreenect2/src/opencl_depth_packet_processor.cpp:50:
/usr/include/CL/cl_version.h:22:104: note: ‘#pragma message: cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)’
   22 | cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)")
      |                                                                                       ^

In file included from /usr/include/CL/opencl.h:26:
/home/amd/libfreenect2/src/opencl_depth_packet_processor.cpp: In constructor ‘libfreenect2::OpenCLDepthPacketProcessorImpl::OpenCLDepthPacketProcessorImpl(int)’:
/home/amd/libfreenect2/src/opencl_depth_packet_processor.cpp:254:15: error: expected unqualified-id before numeric constant
  254 |     const int CL_ICDL_VERSION = 2;
      |               ^~~~~~~~~~~~~~~
/home/amd/libfreenect2/src/opencl_depth_packet_processor.cpp: In member function ‘bool libfreenect2::OpenCLDepthPacketProcessorImpl::fill_trig_table(const libfreenect2::protocol::P0TablesResponse*)’:
/home/amd/libfreenect2/src/opencl_depth_packet_processor.cpp:670:29: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  670 |       const uint16_t *it0 = &p0table->p0table0[r * 512];
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/amd/libfreenect2/src/opencl_depth_packet_processor.cpp:671:29: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  671 |       const uint16_t *it1 = &p0table->p0table1[r * 512];
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/amd/libfreenect2/src/opencl_depth_packet_processor.cpp:672:29: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  672 |       const uint16_t *it2 = &p0table->p0table2[r * 512];
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/freenect2.dir/build.make:1031: CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o] Error 1
make[2]: Leaving directory '/home/amd/libfreenect2/build'
make[1]: *** [CMakeFiles/Makefile2:152: CMakeFiles/freenect2.dir/all] Error 2
make[1]: Leaving directory '/home/amd/libfreenect2/build'
make: *** [Makefile:139: all] Error 2`