OpenKinect/libfreenect2 #733: failed to enable power status U1

zinkcious, 2016-10-05:
I follow the install steps for linux, I'm using ubuntu14.04 on TK1.

when I run the final test instruction "./bin/Protonect", I get following error:

`[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:10 with serial 020354743547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to enable power states U1! LIBUSB_ERROR_PIPE Pipe error. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @2:10
failure opening device!`

Anyone knows why does this happen?

HenningJ, 2016-10-05:
Have you tried t do what it tells you:

    [Error] [protocol::UsbControl] failed to enable power states U1! LIBUSB_ERROR_PIPE Pipe error. Try debugging with environment variable: export LIBUSB_DEBUG=3 .

That should give you (and anybody trying to help you) more information.

zinkcious, 2016-10-06:
It tells me to try debugging with environment variable: export LIBUSB. Actually, I don't really understand what it mean, how can I debug the program? I only that the command 
"./bin/Protonect" can run the program, but I don't know how to debug it with environment variable. Can you tell me how?

HenningJ, 2016-10-06:
Before running `./bin/Protonect` create an environment variable called `LIBUSB_DEBUG`with a value of `3`. Typically you do that by running the command it told you before running protonect, so:

    export LIBUSB_DEBUG=3
    ./bin/Protonect

zinkcious, 2016-10-19:
I export LIBUSB_DEBUG=3 and run the program again, it shows the same error.

By the way, when I use another tk1(I'm installing the driver on tk1) and try install libfreenect2 again, it shows different error but partly same. The new error message is like follows:
`[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:113 with serial 020354743547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_BUSY Resource busy. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @2:113
`
I think it might be because of some problem with tk1, because there is only 1 usb port on tk1, so I usb an usb hub, which support usb3.0.

Sorry I gap a long time before answer you again, I was trapped with other issues. Do you know why this happen?

zinkcious, 2016-10-19:
I have just install a package so that tk1 can enable usb3.0, and after that I run ./bin/Protonect and receive the same error as before, failed to enable power status U1

zinkcious, 2016-10-23:
unsolved, close and open a new one with more info

closed 2016-10-23 by zinkcious