OpenKinect/libfreenect2 #737: Dropping packets on Jetson TX1

natejgardner, 2016-10-15:
Overview Description:
I'm getting extremely low frame rates on Ubuntu 16.04 on the Jetson TX1. Dropping lots of packets. In Protonect, the frame rate is aroun 15 fps, when I launch in iai_kinect, at full resolution, the framerates go down to 1-2 fps. Rtabmap completely refuses to try mapping.

Version, Platform, and Hardware Bug Found:

1. `9fabdc7 Merge pull request #734 from RealRecon/fix_cmake`
2. `Linux tegra-ubuntu 3.10.96-tegra #1 SMP PREEMPT Wed Sep 28 17:51:08 PDT 2016 aarch64 aarch64 aarch64 GNU/Linux`
3. `/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=tegra-ehci/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
    |__ Port 2: Dev 8, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 1: Dev 10, If 0, Class=Hub, Driver=hub/1p, 5000M
            |__ Port 1: Dev 11, If 0, Class=Vendor Specific Class, Driver=, 5000M
            |__ Port 1: Dev 11, If 1, Class=Vendor Specific Class, Driver=, 5000M
            |__ Port 1: Dev 11, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 1: Dev 11, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 2: Dev 9, If 0, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 2: Dev 9, If 1, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/5p, 480M
    |__ Port 3: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 8, If 0, Class=Hub, Driver=hub/1p, 480M
        |__ Port 3: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M
`
4. ``

Steps to Reproduce:

1. `protonect`

Actual Results:
    
    [Debug] [DepthPacketStreamParser] not all subsequences received 831
    [Info] [TegraJpegRgbPacketProcessor] avg. time: 5.00813ms -> ~199.676Hz
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Info] [DepthPacketStreamParser] 2 packets were lost
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [DepthPacketStreamParser] skipping depth packet
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [DepthPacketStreamParser] skipping depth packet
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [RgbPacketStreamParser] skipping rgb packet!
    [Debug] [DepthPacketStreamParser] skipping depth packet
[... 377 lines cut ...]

Additional Information:
Overview Description:
I'm getting extremely low frame rates on Ubuntu 16.04 on the Jetson TX1. Dropping lots of packets. In Protonect, the frame rate is aroun 15 fps, when I launch in iai_kinect, at full resolution, the framerates go down to 1-2 fps. Rtabmap completely refuses to try mapping.

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`: 9fabdc7 Merge pull request #734 from RealRecon/fix_cmake
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. `protonect`
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

xlz, 2016-10-16:
`Protonect cuda`?

natejgardner, 2016-10-16:
`protonect cuda` had the same results. I haven't tried `protonect cuda -noviewer` yet, but I'll let you know as soon as I test that.

xlz, 2016-10-16:
Next time use top -d1 visually to report per thread usage (H thread view, V tree view, I Irix mode to report per core usage).

natejgardner, 2016-10-20:
```
ubuntu@tegra-ubuntu:~/src/libfreenect2$ protonect cuda -noviewer
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: NVIDIA Tegra X1 @ 72MHz Memory 3993MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:13 with serial 089837740747
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[[Debug] [DepthPacketStreamParser] not all subsequences received 512
Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Debug] [DepthPacketStreamParser] subpacket too large
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Debug] [DepthPacketStreamParser] not all subsequences received 192
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Debug] [DepthPacketStreamParser] not all subsequences received 896
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[... 134 lines cut ...]
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[... cut ...]
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
```

xlz, 2016-10-20:
Certain USB problem. I'll look into it.

xlz, 2016-11-02:
I believe this is an issue with registration being too slow. If you run `Protonect cuda -noviewer -norgb` or  `Protonect cuda -noviewer -nodepth` individually, there would be no packet loss.

xlz, 2016-12-05:
Merge into #744.

closed 2016-12-05 by xlz