OpenKinect/libfreenect2 #738: libfreenect2-openni2, depth stream only

EnvnHash, 2016-10-18:
I'm stuck with a quite simple problem: The Kinect v2 seems to have no option no turn off Auto-Exposuring. If the rgb-stream is enabled and the surrounding is to dark, framerates drop dramatically due to longer exposure times. The simple solution is to only use the depth-stream if the rgb-stream is not needed. With libfreenect2 itself this is no problem, but the Openni-Driver seems to activate all streams on default.

Is there any possibility to have the Openni-Driver use a specific stream only?

thanks and cheers!

xlz, 2016-10-19:
Not yet. There is `startStreams()` but the openni2 driver has yet to use it. @hanyazou

EnvnHash, 2016-10-19:
I tried myself on a quick and dirty fix on this issue. Since the whole oni-device mechanism seems to be a bit tricky in initializing I added two environmental variables "LIBFREENECT2_OPENNI_NORGB" and "LIBFREENECT2_OPENNI_NODEPTH". Not very elegant, I know, but works for the moment. 

(attached a patch).

cheers!

[openni_device_patch.txt](https://github.com/OpenKinect/libfreenect2/files/539199/openni_device_patch.txt)

hanyazou, 2016-10-22:
@EnvnHash Sorry for my late reply. And I don't have any Kinect2 device to try with for now. I've investigated your patch, OpenNI2 source code and the driver however. I'm not sure but this small patch might be worth for you.

This does not work with the NiViewer because the NiViewer try to open all types of stream. But I guess your program open the depth stream only. 

[openni_device_patch (2).txt](https://github.com/OpenKinect/libfreenect2/files/545700/openni_device_patch.2.txt)

> diff --git a/src/openni2/DeviceDriver.cpp b/src/openni2/DeviceDriver.cpp
index 3267be4..944a7a1 100644
--- a/src/openni2/DeviceDriver.cpp
+++ b/src/openni2/DeviceDriver.cpp
@@ -59,6 +59,8 @@ namespace Freenect2Driver
     bool device_used;
     libfreenect2::SyncMultiFrameListener listener;
     libfreenect2::thread* thread;
\+    bool enable_rgb;
\+    bool enable_depth;

>     static void static_run(void* cookie)
     {
@@ -172,7 +174,9 @@ namespace Freenect2Driver
       device_stop(true),
       device_used(false),
       listener(libfreenect2::Frame::Depth | libfreenect2::Frame::Ir | libfreenect2::Frame::Color),
\-      thread(NULL)
\+      thread(NULL),
\+      enable_rgb(false),
\+      enable_depth(false)
     {
     }
     ~DeviceImpl()
@@ -205,7 +209,7 @@ namespace Freenect2Driver
         device_used = true;
         device_stop = false;
         thread = new libfreenect2::thread(&DeviceImpl::static_run, this);
\-        dev->start();
\+        dev->startStreams(enable_rgb, enable_depth);
       }
     }
     void stop() {
@@ -249,9 +253,11 @@ namespace Freenect2Driver
           return NULL;
         case ONI_SENSOR_COLOR:
           WriteMessage("Device: createStream(color)");
\+          enable_rgb = true;
           return color;
         case ONI_SENSOR_DEPTH:
           WriteMessage("Device: createStream(depth)");
\+          enable_depth = true;
           return depth;
         case ONI_SENSOR_IR:
           WriteMessage("Device: createStream(ir)");

SirDifferential, 2016-12-14:
We experienced this problem with our project. The solution we implemented was a new FrameListener that doesn't try to synchronize frames. Instead, it always waits only for the depth stream which is always close to 30 FPS, and RGB stream is read only if one arrived independently of the depth stream. It's not really fit for a pull request as it has some things specific to our use-case, but this kind of solution allows you to get depth at 30 FPS when color is enabled. The OpenNI2 `waitForAnyStream()` call does not seem to only wait for Depth if you ask it to ignore color.