OpenKinect/libfreenect2 #741: failed to enable power status U1! LIBUSB_ERROR_PIPE Pipe error.

zinkcious, 2016-10-23:
I'm using tk1, linux4tegra 21.4. I'm trying to install libfreenect2 on tk1 to get the kinect2 data. I step by step install the packages, but when I run ./bin/Protonect to check the correctness of the installation, I encounter following errors:
`
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 devices connected
[Info] [Freenect2Impl] Found valid Kinect v2 @1:15 with serial 020354743574
[Info] [Freenect2Impl] Found 1 devices
[Info] [Freenect2DeviceImpl] close
[Error] [protocol::UsbControl] failed to enable power status U1! LIBUSB_ERROR_PIPR Pipe error. Try debuging with environment variabl: export LIBUSB_DEBUG=3
[Info] [Freenect2DeviceImpl] closing 
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @1: 15
failure opening device!
`

I export LIBUSB_DEBUG=3 and ./bin/Protonect again and get exactly the same info.

I'm using a usbhub with usb3.0, when I type in lsusb -t in the terminal, I get follows:
`
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=tegra-ehci/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/2p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/6p, 480M
    |__ Port 2: Dev 2, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 2: Dev 2, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 3: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 9, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 1: Dev 9, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 6, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 14, If 0, Class=Hub, Driver=hub/1p, 480M
                |__ Port 1: Dev 15, If 0, Class=Vendor Specific Class, Driver=, 480M
                |__ Port 1: Dev 15, If 1, Class=Audio, Driver=, 480M
                |__ Port 1: Dev 15, If 2, Class=Audio, Driver=, 480M
`

When I type in "lspci |grep USB", I get nothing back
When I type in "lspci", I get following:
`
00:00.0 PCI bridge: NVIDIA Corporation TegraK1 PCIe x4 Bridge (rev a1)
01:00.0 Network controller: Intel Corporation Wireless 7260 (rev bb)
02:00.0 PCI bridge: NVIDIA Corporation TegraK1 PCIe x1 Bridge (rev a1)
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)
`

Anyone knows why I encounter such problem? If you want more info, please leave a message here, I will response as quickly as possible. I'm stuck here for a long time, I will really appreciate it very much if you can help me. Any possible solution will help, I just don't know what to do now. Thanks!

xlz, 2016-10-24:
Don't use a usbhub.

zinkcious, 2016-10-26:
But the tk1 has only one usb port, if I don't use a usbhub, I can't usb other usb devices such a laser. 

And I tried what you said, don't use a usbhub, plug the usb port of kinect directly into the usb port on tk1 and remote control tk1. But I encounter new problems when I do this.

If I connect tk1 using ssh, and input ./bin/Protonect, it tells me following errors

> [Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 6 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @1:16 with serial 020354743547
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65544 X11: The DISPLAY environment variable is missing
[Error] [OpenGLDepthPacketProcessor] Failed to initialize GLFW.

If I connect tk1 using remote desktop, and input ./bin/Protonect, it tells me following errors

> [Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 6 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @1:16 with serial 020354743547
[Info] [Freenect2Impl] found 1 devices
X Error of failed request: BadRequest (invalid request code or no such operation)
    Major opcode of failed request 149(RANDR)
    Minor opcode of failed request 8 (RRGetScreenResources)
    Serial number of failed request: 14
    Currenr serial number in output stream: 14

What any other suggestions fo you have?

xlz, 2016-10-26:
`protonect cuda`

zinkcious, 2016-10-27:
I try your suggestion "protonect cuda -noviewer" in three condition as follows:

1. kinect2 connect to tk1 directly via usb, ssh remote login tk1
It shows no more error like "GLFW error 65544 X11: The DISPLAY environment variable is missing", but it shows error "failed to enable power states U1! LIBUSB_ERROR_PIPE Pipe error", which is the error I have encountered before.

2. kinect2 connect to tk1 directly via usb, remote desktop login tk1
the error is exactly the same as I input "./bin/Protonect", which is "X error of failed request: Badrequest (invalid request operation or no such operation)"

3. kinect2 connect to a usbhub, which is also connected to mouse and keyboard so I can control the tk1. And usbhub connect to tk1.
the error is same before "failed to enable power states U1! LIBUSB_ERROR_PIPE Pipe error"

Do you have any other suggestions? Or can you analyse what may be the reason that cause this problem? Thanks a lot!

xlz, 2016-10-27:
I've never seen LIBUSB_ERROR_PIPE if you connect kinect2 directly, so no idea about this.

"protonect cuda -noviewer" doesn't use X at all so X errors are impossible.

I have no other suggestions.

xlz, 2016-12-05:
I believe this is  CANTREPRODUCE.

closed 2016-12-05 by xlz

saiwael, 2019-05-09:
Hello, I had this error while launching ./bin/Protonect, on ubuntu 14.04 I did not understand the origin of the problem: it is a problem related with libusb or it is a problem of the pc? My problem is that it already worked before and I can see rgb and depth images
Thank you in advance.
saidane@saidane-Inspiron-3521:~/libfreenect2/build$ ./bin/Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:29 with serial 009074163547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 64*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_PIPE Pipe error
[Info] [Freenect2DeviceImpl] closing...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_PIPE Pipe error
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed