OpenKinect/libfreenect2 #743: libusb errors while following step-by-step instruction from Readme(ubuntu 14.04)

pranali3215, 2016-10-24:
Hi,

I have been following the step-by-step instructions for linux(ubuntu 14.04) but I have been getting the following error on running the cmake command:

-- using tinythread as threading library
-- checking for module 'libusb-1.0>=1.0.20'
--   package 'libusb-1.0>=1.0.20' not found
CMake Error at /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:283 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:337 (_pkg_check_modules_internal)
  cmake_modules/FindLibUSB.cmake:21 (PKG_CHECK_MODULES)
  CMakeLists.txt:93 (FIND_PACKAGE)

-- checking for modules 'libva;libva-drm'
--   package 'libva' not found
--   package 'libva-drm' not found
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Could NOT find OpenCL (missing:  OpenCL_INCLUDE_DIR) 
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- Linking with these libraries: 
 LibUSB_LIBRARY-NOTFOUND
 pthread
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /usr/lib/x86_64-linux-gnu/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OpenNI2_LIBRARY_DIR OpenNI2_INCLUDE_DIRS) 
-- Feature list:
--   CUDA    no
--   CXX11    disabled
--   Examples    yes
--   OpenCL    no
--   OpenGL    yes
--   OpenNI2    no
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LibUSB_LIBRARY
    linked by target "freenect2" in directory /home/pranali/Downloads/libfreenect2

-- Configuring incomplete, errors occurred!
See also "/home/pranali/Downloads/libfreenect2/build/CMakeFiles/CMakeOutput.log".
See also "/home/pranali/Downloads/libfreenect2/build/CMakeFiles/CMakeError.log".

I also checked the libusb versin using the command: dpkg -l libusb-1.0-*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                 Version                 Architecture            Description
+++-====================================-=======================-=======================-==============================================================================
ii  libusb-1.0-0:amd64                   2:1.0.20-1ppa1~trusty1  amd64                   userspace USB programming library
ii  libusb-1.0-0-dev:amd64               2:1.0.20-1ppa1~trusty1  amd64                   userspace USB programming library development files
un  libusb-1.0-doc                       <none>                  <none>                  (no description available)

Any help would be appreciated.

xlz, 2016-10-24:
Have you built libusb from source anywhere?

pranali3215, 2016-10-24:
No. I initially had an old version running but I removed it and just ran the instructions in the Readme.

xlz, 2016-10-24:
`pkg-config --debug libusb-1.0`?

closed 2016-10-31 by xlz

pranali3215, 2016-11-14:
Output of pkg-config --debug libusb-1.0:

Looking for package 'libusb-1.0'
Looking for package 'libusb-1.0-uninstalled'
Reading 'libusb-1.0' from file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
Parsing package file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
  line>prefix=/usr/local
 Variable declaration, 'prefix' has value '/usr/local'
  line>exec_prefix=${prefix}
 Variable declaration, 'exec_prefix' has value '/usr/local'
  line>libdir=${exec_prefix}/lib
 Variable declaration, 'libdir' has value '/usr/local/lib'
  line>includedir=${prefix}/include
 Variable declaration, 'includedir' has value '/usr/local/include'
  line>
  line>Name: libusb-1.0
  line>Description: C API for USB device access from Linux, Mac OS X, OpenBSD, NetBSD and Windows userspace
  line>Version: 1.0.9
  line>Libs: -L${libdir} -lusb-1.0
  line>Libs.private: -lrt -pthread
Unknown keyword 'Libs.private' in '/usr/local/lib/pkgconfig/libusb-1.0.pc'
  line>Cflags: -I${includedir}/libusb-1.0
  line>
Path position of 'libusb-1.0' is 1
Adding 'libusb-1.0' to list of known packages, returning as package 'libusb-1.0'

xlz, 2016-11-14:
So you have libusb 1.0.9 installed in /usr/local/lib/. Remove it?

pranali3215, 2016-11-14:
I do an sudo apt-get purge --auto-remove libusb-1.0-0 but it still shows I have Version 1.0.9 installed.

xlz, 2016-11-14:
`find /usr/local -name '*libusb*' -delete`

pranali3215, 2016-11-14:
Thank you for your help and patience !!
It has started working now.

fcastilloc, 2016-11-21:
Hi,
I have a similar problem following the instructions for LinuxMint, when I run cmake command:

-- using tinythread as threading library
-- checking for modules 'libva;libva-drm'
--   package 'libva' not found
--   package 'libva-drm' not found
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
CMake Error at /usr/share/cmake-3.0/Modules/FindPackageHandleStandardArgs.cmake:136 (message):
  Could NOT find TurboJPEG (missing: TurboJPEG_INCLUDE_DIRS TURBOJPEG_WORKS)
Call Stack (most recent call first):
  /usr/share/cmake-3.0/Modules/FindPackageHandleStandardArgs.cmake:343 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindTurboJPEG.cmake:66 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:240 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/home/libfreenect2/build/CMakeFiles/CMakeOutput.log".

ssw123, 2017-03-11:
I have a similar problem 
but  Output of pkg-config --debug libusb-1.0

Looking for package 'libusb-1.0'
Looking for package 'libusb-1.0-uninstalled'
Reading 'libusb-1.0' from file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
Parsing package file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
  line>prefix=/usr/local
 Variable declaration, 'prefix' has value '/usr/local'
  line>exec_prefix=${prefix}
 Variable declaration, 'exec_prefix' has value '/usr/local'
  line>libdir=${exec_prefix}/lib
 Variable declaration, 'libdir' has value '/usr/local/lib'
  line>includedir=${prefix}/include
 Variable declaration, 'includedir' has value '/usr/local/include'
  line>
  line>Name: libusb-1.0
  line>Description: C API for USB device access from Linux userspace
  line>Version: 1.0.0
  line>Libs: -L${libdir} -lusb-1.0
  line>Cflags: -I${includedir}/libusb-1.0
  line>
Path position of 'libusb-1.0' is 2
Adding 'libusb-1.0' to list of known packages, returning as package 'libusb-1.0'

can you help me?
thanks

floe, 2017-03-11:
https://github.com/OpenKinect/libfreenect2/issues/743#issuecomment-260277106

alanibrahimovic, 2017-05-23:
I have a similar error however on a windows machine (Windows 10 64 bit machine)

I have received the following output when running  `cmake .. -G "Visual Studio 12 2013 Win64"`

`Found PkgConfig: C:/MinGW/bin/pkg-config.exe (found version "0.26")
Checking for module 'libusb-1.0'
No package 'libusb-1.0' found
CMake Error at C:/Program Files/CMake/sgare/cmake-3.8/Modules/FindPkgConfig.cmake:416 (message):
A required package was not found
Call Stack (most recent call first):
C:/Program Files/CMake/share/cmake-3.8/Modules/FindPkgConfig.cmake:589 (_pkg_check_modules_internal)
cmake_modules/FindLibUSB.cmake:21 (PKG_CHECK_MODULES)
CMakeLists.txt:93 (FIND PACKAGE)`

I have installed libusbK driver using Zadig including "Scan for hardware changes"
Build libusb with .\install_libusb_vs2015.cmd (my version of vs is 2015)

Could you please help
Thanks in advance

DonDonat, 2017-06-23:
After trying like about everything anyone ever mentioned to solve the problem @alanibrahimovic mentioned before, I have to admit I'm unable to get libfreenect2 up and running on a Win10 machine.

First of all: I have installed every required tool (cmake, [PckConfig via chocolatey](https://chocolatey.org/packages/pkgconfiglite), TurboJPEG, GLFW and even the optional ones OpenCL and OpenNI2) and I tried UsbDk as well as libusbK.

But unimportant what I do, I always get the error @alanibrahimovic mentioned (yes, the same, with the exact same lines).

Like the tutorial said, I went to `cd depends/` and executed
`.\install_libusb_vs2015.cmd` without errors.

Then I created build and entered that folder. After that, I tried `cmake .. -G "Visual Studio 14 2015 Win64"` in GitBash and got the same error as @alanibrahimovic.
To make sure, it is not a mistake with adminestrator right e.g I tried Visual Studio Command Prompt as well as Windows shell and still got the same mistake. Even using the cmake user interface on windows caused the same problem.

I really need that framework to start coding the program required for my diplomma so it would be really nice if someone does know any solutions for that problem...