OpenKinect/libfreenect2 #746: NaN returned from getPointXYZRGB

kiss-lawrence, 2016-10-27:
I'm playing with the sample program Protonect.cpp. The programs runs fine, but when I added a getPointXYZRGB call in the frame process loop to examine the point cloud data, it always return NaN for x, y and z, 0 for RGB. And the format for the registered frame is invalid.

xlz, 2016-10-28:
Can you provide the code you added?

kiss-lawrence, 2016-10-29:
for (int I = 0; I < registered.height; ++I)
    {
        for (int J = 0; J < registered.width; ++J)
        {
            float X, Y, Z, RGBVALUE;
            registration->getPointXYZRGB(&undistorted, &registered, I, J, X, Y, Z, RGBVALUE);
            if (!std::isnan(X) || !std::isnan(Y) || !std::isnan(Z))
            {
                std::cout << " X = " << X 
                            << " Y = " << Y 
                            << " Z = " << Z 
                            << " RGB = " << RGBVALUE
                            << std::endl;
            }
        }
    }

After registration->apply() has been called

xlz, 2016-10-31:
I cannot reproduce this error.

closed 2016-10-31 by xlz

NintyS, 2023-03-18:
I have it too. Do you know what was wrong?