OpenKinect/libfreenect2 #747: Getting Raw data

manideepabc, 2016-10-28:
Hi ,

I am trying to get raw data(phase images) from kinectv2 using libfreenect2. I tried to create a DumpPacketPipeline and attached it the device. I have also set Ir and Depth frame listeners.
When I run the program I get segmentation fault(core dumped) in the terminal.

I am not sure how to do this. Can someone show the code for getting raw data?

What is the difference between Ir and Depth frames?
Thanks in advance.

xlz, 2016-10-28:
Can you provide your code of DumpPacketPipeline?

3 phases at 3 frequencies. 3x3=9.

Ir is amplitude.

manideepabc, 2016-10-28:
Thank you for your reply.
I used the following code
`pipeline = new libfreenect2::OpenCLPacketPipeline(deviceId);`
  `if(freenect2.enumerateDevices() == 0)`
  `{`
    `std::cout << "no device connected!" << std::endl;`
    `return -1;
  `}`

  `if (serial == "")`
  `{`
    `serial = freenect2.getDefaultDeviceSerialNumber();`
  `}`

  `if(pipeline)`
  `{`
    `dev = freenect2.openDevice(serial, pipeline);` 
  `}`
  `else`
  `{`
    `dev = freenect2.openDevice(serial);`
  `}`

`if (enable_depth){`
  	`types |= libfreenect2::Frame::Ir;`
  	`types |= libfreenect2::Frame::Depth;`
  }
    
   _I have set the listener like this_
 
  `libfreenect2::SyncMultiFrameListener listener(types);`
  `libfreenect2::FrameMap frames;`

  `dev->setIrAndDepthFrameListener(&listener);`

_To get the frames I used_ 

`libfreenect2::Frame *rgb = frames[libfreenect2::Frame::Color];`
    `libfreenect2::Frame *ir = frames[libfreenect2::Frame::Ir];`
    `ir->format = libfreenect2::Frame::Raw;`
    `libfreenect2::Frame *depth = frames[libfreenect2::Frame::Depth];`
    `depth->format = Raw;`

Also does Kinect's hardware uses three modulation frequencies for each frame? Is this anyway related to three sources of light present on the Kinect?

xlz, 2016-10-31:
You were accessing `rgb` without specifying RGB to be provided as part of the output.

closed 2016-10-31 by xlz

sairamkolla, 2017-03-02:
@xlz How to save the raw depth to some directory? 

I looked up in the documentation but couldn't find it anywhere.