OpenKinect/libfreenect2 #748: i965 libva does not work

Tobias-Fischer, 2016-11-01:
Overview Description:
When trying to use the i965 VA-API driver, I get a `[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error`.

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
`83f88b4 Update author list for 0.2 release`
2. `uname -a`
`Linux deepthought 4.4.0-45-generic #66~14.04.1-Ubuntu SMP Wed Oct 19 15:05:38 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`
3. `lsusb -t`
```
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 3: Dev 3, If 0, Class=Hub, Driver=hub/2p, 480M
        |__ Port 1: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 1: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 1: Dev 5, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 4: Dev 4, If 0, Class=Hub, Driver=hub/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 6: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
```
4. `lspci -nn`
```
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor DRAM Controller [8086:0150] (rev 09)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor PCI Express Root Port [8086:0151] (rev 09)
00:02.0 Display controller [0380]: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor Graphics Controller [8086:0162] (rev 09)
00:14.0 USB controller [0c03]: Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller [8086:1e31] (rev 04)
00:19.0 Ethernet controller [0200]: Intel Corporation 82579LM Gigabit Network Connection [8086:1502] (rev 04)
00:1a.0 USB controller [0c03]: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #2 [8086:1e2d] (rev 04)
00:1b.0 Audio device [0403]: Intel Corporation 7 Series/C210 Series Chipset Family High Definition Audio Controller [8086:1e20] (rev 04)
00:1c.0 PCI bridge [0604]: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 1 [8086:1e10] (rev c4)
00:1c.4 PCI bridge [0604]: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 5 [8086:1e18] (rev c4)
00:1d.0 USB controller [0c03]: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #1 [8086:1e26] (rev 04)
00:1e.0 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev a4)
00:1f.0 ISA bridge [0601]: Intel Corporation Q77 Express Chipset LPC Controller [8086:1e47] (rev 04)
00:1f.2 SATA controller [0106]: Intel Corporation 7 Series/C210 Series Chipset Family 6-port SATA Controller [AHCI mode] [8086:1e02] (rev 04)
00:1f.3 SMBus [0c05]: Intel Corporation 7 Series/C210 Series Chipset Family SMBus Controller [8086:1e22] (rev 04)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:1c82] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:0fb9] (rev a1)
03:00.0 Ethernet controller [0200]: Intel Corporation 82574L Gigabit Network Connection [8086:10d3]
```

Steps to Reproduce:
1. Follow installation instructions, including libva dependencies for Ubuntu 14.04
2. Build libfreenect with `ENABLE_VAAPI=ON`
3. Run `./bin/Protonect`

Actual Results:
```
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
```

Expected Results:
libva initialized properly

Additional Information:
Output of `vainfo`:
```
vainfo
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_0_32
libva info: va_openDriver() returns 0
vainfo: VA-API version: 0.39 (libva 1.7.0)
vainfo: Driver version: Splitted-Desktop Systems VDPAU backend for VA-API - 0.7.3
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG4Simple            :	VAEntrypointVLD
      VAProfileMPEG4AdvancedSimple    :	VAEntrypointVLD
      VAProfileH264Baseline           :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
```

Output of `LIBVA_DRIVER_NAME=i965 vainfo`
```
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
vainfo: intel_driver.c:100: intel_driver_init: Assertion `(((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_DRI1)) || (((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_DRI2)) || (((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_CUSTOM))' failed.
Aborted (core dumped)
```
So I guess it is not an issue of libfreenect2, but of the libva Intel driver. I have an IvyBridge CPU, so libva should be working (Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz).

Do you have any hints on how to approach this?

Thanks, Tobias

xlz, 2016-11-01:
Either uninstall vdpau backend of libva or use `LIBVA_DRIVER_NAME=i965 Protonect`.

closed 2016-11-01 by xlz

Tobias-Fischer, 2016-11-01:
I have the same issue as with `vainfo` when launching `LIBVA_DRIVER_NAME=i965 ./bin/Protonect`:

```
tobias@deepthought:~/robot/tools/libfreenect2/build$ LIBVA_DRIVER_NAME=i965 ./bin/Protonect
./bin/Protonect: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/tobias/robot/tools/libfreenect2/build/lib/libfreenect2.so.0.2)
./bin/Protonect: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/tobias/robot/tools/libfreenect2/build/lib/libfreenect2.so.0.2)
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 038899543547
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
Protonect: intel_driver.c:100: intel_driver_init: Assertion `(((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_DRI1)) || (((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_DRI2)) || (((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_CUSTOM))' failed.
Aborted (core dumped)
```

xlz, 2016-11-01:
It's not the same issue.

> Protonect: intel_driver.c:100: intel_driver_init: Assertion `(((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_DRI1)) || (((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_DRI2)) || (((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_CUSTOM))' failed.
Aborted (core dumped)

This is your drm not being correctly configured.

Since you're using Nvidia/Intel dual graphics, what is working known to me is to use Intel for graphics and Nvidia for compute. `prime-select intel`

curatorcorpus, 2017-07-12:
I have had the same problem when trying to run the vaapi for rgb packets and cuda for depth packets.

I tried prine-select intel but this error shows:
Fatal error in __driConfigOptions line 1, column 0: unknown encoding.
Aborted (core dumped)

Any ideas?

Thanks