OpenKinect/libfreenect2 #749: OpenNI2 driver does not free devices before process exits

SirDifferential, 2016-11-03:
Overview Description:

When using the OpenNI2 driver on Windows or OS X, calling `openni::Device::open()` will claim the USB device and make it unusable by other processes.

Version, Platform, and Hardware Bug Found:

1. My own fork from latest master `5819b69` with a few changes to device enumeration and selecting the pipeline in DeviceDriver.cpp
2. `uname -a`: Darwin DelicodeMBP 15.5.0 Darwin Kernel Version 15.5.0: Tue Apr 19 18:36:36 PDT 2016; root:xnu-3248.50.21~8/RELEASE_X86_64 x86_64 and Windows 8.1 64-bit
3. `lsusb -t`: Not available on the platforms
4. `lspci -nn`: Not available on the platforms

Steps to Reproduce:

* Enumerate all devices by calling `openni::OpenNI::initialize()` followed by `openni::OpenNI::enumerateDevices()`
* Iterate through the found devices and open devices by uri: `device.open(dev_info.getUri())`, then close them by `device.close()`.
* Keep the process running
* In another process, perform the same steps

Actual Results:

In the second process:
    * On Windows, 0 devices are found
    * On OS X, a crash happens in libfreenect2.cpp at `Freenect2Device *Freenect2::openDevice(int idx, const PacketPipeline *pipeline, bool attempting_reset)` when the device is being reseted.

Expected Results:

Calling openni::Device::stop() is expected to free the USB device and make it available for other processes, but this does not happen.

Reproducibility:

Happens on Windows 8.1 and OS X 10.11 every time.

Additional Information:

The temporary hack I created for this is by changing device enumeration so that OpenNI2 only ever gets one device from the enumeration, even if multiple are connected:

* `libfreenect2.cpp::enumerateDevices()` reads the currently used serial numbers from an env var I set before calling `openni::OpenNI::initialize()`
* It adds only the first free sensor to the enumerated devices container, making sure only one sensor is ever enumerated.

This hack avoids the problem. I'm testing with 2 Kinects. Using libfreenect2 directly without OpenNI2 works.

I'm assuming the problem is related to the `DeviceDriver.cpp` function `DeviceImpl::stop()` which does not correctly free some OpenNI2 resources before the process exits.

hanyazou, 2016-11-03:
Could you please share your code or patch here?

SirDifferential, 2016-11-03:
https://github.com/SirDifferential/libfreenect2_openni2_tester

This repo contains a minimal example that causes a crash on OS X, using the vanilla master branch without any edits. It includes a pre-compiled binary (`a.out`) and the libs for OpenNI2 and libfreenect2. It can be tested as follows:

* Connect 2 Kinects
* Run `a.out` in terminal 1
* Make sure it opens a device and enters the eternal sleep loop
* Run `a.out` in terminal 2, crash happens

As for the hack to circumvent this, it's really a hack and probably not worth submitting - all it does is expose only a single device in each process when OpenNI2 calls `enumerateDevices()`

xlz, 2016-11-03:
Can this be reproduced with libfreenect2 only?

SirDifferential, 2016-11-03:
Removing OpenNI2 from the test and trying this using only base libfreenect2 works fine without any issues.

SirDifferential, 2016-11-04:
For verbosity's sake, here's a tester without openni: https://github.com/SirDifferential/libfreenect2_openni2_tester/blob/master/main_noopenni.cpp

It outputs the following:

```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:20 with serial 501543341942
[Info] [Freenect2Impl] found valid Kinect v2 @20:17 with serial 002112160847
[Info] [Freenect2Impl] found 2 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 4*128*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] already stopped, doing nothing
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
opened and closed device
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 4*128*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] already stopped, doing nothing
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
opened and closed device
```

xlz, 2016-11-04:
There are some debug messages in openni2 code that can help https://github.com/OpenKinect/libfreenect2/blob/master/src/openni2/DeviceDriver.cpp#L212

Try setting openni2 logging level to debug?

xlz, 2016-11-04:
There are some subtle state machine stuff going on with close() and stop(). Probably need to take a close look at it.

hanyazou, 2016-11-05:
@xls you can enable that debug messages putting these lines to libfreenect2_openni2_tester/OpenNI2_lib/OpenNI.ini

[Log]
; 0 - Verbose; 1 - Info; 2 - Warning; 3 - Error. Default - None                                                                                                                                                                                                    
Verbosity=0
LogToConsole=1

hanyazou, 2016-11-05:
the DeviceImpl::close() in DeviceDriver.cpp does not call libfreenect2::Freenect2Device::close() if it is called without calling start().
https://github.com/OpenKinect/libfreenect2/blob/master/src/openni2/DeviceDriver.cpp#L221

If you call openni::Device::close() without any streaming, the driver won't call close() and keep the libfreenect2 instance until the end of process.

#574 introduce this problem.
> This might introduce some memory leak. But it is better than hangup.

xlz, 2016-11-07:
So in term of this bug, a workaround would be to insert a call to `start()` between `open()` and `close()`.

But I think a proper fix is to add a pair of `start()` and `stop()` in `Freenect2Driver::DeviceImpl::close()` when the OpenNI device is not started once. @hanyazou?

xlz, 2016-11-07:
OK I get it now. #574 is introducing the design to prevent turning Kinect on and off repeatedly and this is reasonable. 

The report claims Expected Results as

> Calling openni::Device::stop() is expected to free the USB device and make it available for other processes, but this does not happen.

The design of not turning it on and off repeatedly means you can't transfer access of Kinect between processes. This is also how Microsoft does it - it integrates all hardware access in one process -KinectService.exe.

I believe each device will have their own thread so there shouldn't be any issue with multiple Kinects in one process. So this isn't actually the expected behavior.

Can I ask what you want to achieve by opening all devices and closing them immediately after? @SirDifferential

SirDifferential, 2016-11-08:
In the OpenNI2 driver the sensor serial number is masked: After devices are enumerated, checking the assorted values in openni::DeviceInfo structs produces "Microsoft Kinect" as the serial number (to make the driver work with NiTE2 I assume). Every connected sensor will have this identical serial number. It's possible to make a call to the underlying `libfreenect2::Freenect2Device" object by calling openni::Device::getProperty() and then add a line in DeviceDriver.cpp::getPoperty() that returns the serial number. As far as I know, it's not possible to request the serial number via OpenNI2 without first opening the device.

The serial numbers can be found via libfreenect2 directly (using enumeration), but I'm not sure if this is guaranteed to match to the OpenNI devices one-to-one (will libfreenect2 enumerated_devices container have the same device at index n as the container for OpenNI2 devices?)

xlz, 2016-11-08:
A better solution would be adding serial number to the uri in https://github.com/OpenKinect/libfreenect2/blob/master/src/openni2/DeviceDriver.cpp, like `freenect2://0?depth-size=123x456&serial=0123456789` and prefer serial number than device id.

SirDifferential, 2016-11-08:
I agree, that's a better way of doing it and we can achieve the functionality we require by using the changed URIs. After I add these changes to my fork, would you like a pull request for it or is it something trivial that's not considered relevant for master?

HenningJ, 2016-11-08:
> A better solution would be adding serial number to the uri in https://github.com/OpenKinect/libfreenect2/blob/master/src/openni2/DeviceDriver.cpp, like freenect2://0?depth-size=123x456&serial=0123456789 and prefer serial number than device id.

Or rather `freenect2://0123456789?depth-size=123x456`, foregoing the device id completely. I actually have a branch lying around somewhere implementing that.

xlz, 2016-11-08:
> Or rather freenect2://0123456789?depth-size=123x456

True if nobody has anything that depends on the URI containing device id.

xlz, 2016-11-08:
I have no idea about what OpenNI2 API says about the URI. The thing with freenect2://0123456789 is that there is no guarantee about what the serial number would look like to distinguish from device id.

I think a while ago I saw someone implement the `ONI_DEVICE_PROPERTY_SERIAL_NUMBER`, which is not implemented in `Device::getProperty()` here so you couldn't get it. This would be the fix without changing any semantics. Adding `freenect2://0123456789` would apparently make things easier but you have to make sure it is okay with OpenNI2 API.

HenningJ, 2016-11-10:
OpenNI says:

> The URI string format is determined by the driver

So we can do whatever we want. :-)
I think it should just be the serial number, removing the device id (since the device id is not really a meaningful number, as it can change when you add or remove _another_ device). 

I also think the `depth-size` part should be removed as well. The OpenNI way to change resolution is to [set a different video mode](https://github.com/occipital/OpenNI2/blob/master/Include/OpenNI.h#L949) for a stream, not by opening different devices.

xlz, 2016-11-10:
So make it the serial number then.

`depth-size` is likewise a shorthand to set video mode. It can stay.

xlz, 2016-12-05:
Since actually implementing `freenect2://0123456789` is harder than I thought, only `?serial=0123456789` is implemented for informative purposes.

closed 2016-12-05 by xlz