OpenKinect/libfreenect2 #752: Bandwidth requested by KinectV2

sdenenberg, 2016-11-14:
Hello,

I am running a system that requires using multiple kinects in parallel.  Currently I am using 3 USB 3.0 controllers to control 3 kinects.  I understand that you cannot plug two kinects into the same controller because the kinect requests 3GB/s of bandwidth and each controller has only 5 GB/s.  Would it be possible to reduce the bandwidth that is requested?  What bandwidth is actually necessary? I'm not using the RGB data - does this help?

Thanks,

Scott

hanshammel1337, 2016-11-15:
You can find some bandwidth infos here: https://github.com/OpenKinect/libfreenect2/issues/320#issuecomment-150756072

floe, 2016-11-15:
The bandwidth request is done by the firmware, so not much influence from our side. The RGB data is nearly negligible compared to the depth data, so that also won't help much.

sdenenberg, 2016-11-15:
Why does the Kinect request so much bandwidth if it doesn't need it?  There is no way to trick the system into only allocating a smaller bandwidth so that we can use 2 kinects per PCI slot?

Thanks,

Scott

floe, 2016-11-16:
I have a vague recollection that some USB3 host controllers will allow you to "overcommit" bandwidth, i.e. allow devices to request more bandwidth in total than would theoretically be available. Maybe this can be enabled by using one of the "quirk" parameters of the `xhci-hcd` module, but be aware that this is outside of the USB3 spec and will very likely not work...

sdenenberg, 2016-11-16:
I am more than happy to be the guinea pig and report my results.  I imagine that many people would find this useful if it were possible to trick a usb controller into controlling more than one kinect.

Do you know how to modify these "quirk" parameters?  I'm about to start looking into it and any guidance would be appreciated.  Do you have any gut feelings as to why it would very likely not work?

Thank you for your responsiveness.  The support for libfreenect2 has been tremendous.

xlz, 2016-11-16:
> Why does the Kinect request so much bandwidth if it doesn't need it?

It needs it. The isochronuous data has realtime requirement. If it's not transferred in time it's dropped. That's why you have to have exclusive bandwidth allocation and can't have two devices compete for bandwidth.

closed 2016-11-16 by xlz

floe, 2016-11-17:
For the record, the only "documentation" about the quirks I know is the actual kernel source code: http://lxr.free-electrons.com/source/drivers/usb/host/xhci.h#L1621