OpenKinect/libfreenect2 #755: registration frame data lose

Abbyls, 2016-11-18:
Overview Description:
environment: ubuntu 14.04 x86_64
i think my registration frame get from Protonect.cpp have lose many useful data(i think it's the depth frame problem) campared with the registration frame get from the kinect1(driver:libfreenect), and it affect my work, i don't kown if there any method or parameters to set the kinect2 to improve the results.

1. ./bin/Protonect cuda with my kinect2 the frame as follows:
[image]

2. the point cloud data get from the kinect1 as:
[image]

You can see, the kinect2 registration frame can't get the white line in the rgb frame, the data is losed in the depth frame, so, i can't get the information, but in the point cloud from kinect1, we can see the white line and have a more broad sight.

I wonder what's probelm? If i need change some parameters?

Abbyls, 2016-11-18:
add a more clear image from kinect2 Protonect:
[image]

floe, 2016-11-21:
Well, the Kinect1 and Kinect2 are very different cameras, you can't say that one is generally better than the other. Maybe the Kinect1 is just better suited for your specific application.

Abbyls, 2016-11-23:
ok, after that, it's because i didn't remove the protect membrane on the new kinect2!
but the depth image also not well as you can see (1) and  i test the depth_image from the iai_kinect2/kinect2_bridge, as (2),
(1)
[image]
(2)
[image]

You can see, it's different. and i find the kinect2_bridge get headerfiles from freenect package, but protonect get header files from libfreenect2/include! 

Then i copied the header files from freenect to the libfreenect2/include errors happened. 
How can i get depth frame as the kinect2_bridge?

Abbyls, 2016-11-23:
I solved it. As the kinect2_bridge have a config on the device.  You can closed the problem.

closed 2016-11-23 by Abbyls