OpenKinect/libfreenect2 #756: cmake error

fcastilloc, 2016-11-21:
Hi,

I'm following the steps to install libfreenect2 on Linux, but when I make: 
 cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2
I got:
-- using tinythread as threading library
-- checking for modules 'libva;libva-drm'
--   package 'libva' not found
--   package 'libva-drm' not found
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
CMake Error at /usr/share/cmake-3.0/Modules/FindPackageHandleStandardArgs.cmake:136 (message):
  Could NOT find TurboJPEG (missing: TurboJPEG_INCLUDE_DIRS TURBOJPEG_WORKS)
Call Stack (most recent call first):
  /usr/share/cmake-3.0/Modules/FindPackageHandleStandardArgs.cmake:343 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindTurboJPEG.cmake:66 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:240 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/home/charles/Downloads/libfreenect2/build/CMakeFiles/CMakeOutput.log".

xlz, 2016-11-22:
Your turbojpeg isn't set up correctly.

fcastilloc, 2016-11-22:
How can I set it up correctly?

xlz, 2016-11-22:
`turbojpeg.h` has to be located in system include paths.

fcastilloc, 2016-11-22:
Finally it works,
thanks.

closed 2016-11-22 by fcastilloc

sram022, 2018-02-08:
I have the same problem but not sure how to resolve it.How to setup turbojpeg correctly in Ubuntu.