OpenKinect/libfreenect2 #758: How to upgrade OpenGL version more than 3.1 on Ubuntu 14.04, Intel Corporation Sky Lake Integrated Graphics

iaiman, 2016-11-23:
Sorry.
Please teach me how to upgrade OpenGL version more than 3.1?
I’m trying install libfreenect2 on Ubuntu 14.04.
PC(Intel inspiron 13 7000) environment is as below.

rd1gros@rd1gros:~/libfreenect2/build$ uname -r
4.2.0-42-generic

rd1gros@rd1gros:~/libfreenect2/build$ glxinfo |grep OpenGL
libGL error: pci id for fd 4: 8086:1916, driver (null)
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Skylake ULT GT2 
OpenGL version string: 3.0 Mesa 10.5.9
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:

rd1gros@rd1gros:~$ ls /usr/include/GLFW
glfw3.h  glfw3native.h

rd1gros@rd1gros:~$ lspci
00:00.0 Host bridge: Intel Corporation Sky Lake Host Bridge/DRAM Registers (rev 08)
00:02.0 VGA compatible controller: Intel Corporation Sky Lake Integrated Graphics (rev 07)
00:04.0 Signal processing controller: Intel Corporation Device 1903 (rev 08)
00:13.0 Non-VGA unclassified device: Intel Corporation Device 9d35 (rev 21)
00:14.0 USB controller: Intel Corporation Device 9d2f (rev 21)
00:14.2 Signal processing controller: Intel Corporation Device 9d31 (rev 21)
00:15.0 Signal processing controller: Intel Corporation Device 9d60 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Device 9d61 (rev 21)
00:16.0 Communication controller: Intel Corporation Device 9d3a (rev 21)
00:17.0 SATA controller: Intel Corporation Device 9d03 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Device 9d14 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Device 9d48 (rev 21)
00:1f.2 Memory controller: Intel Corporation Device 9d21 (rev 21)
00:1f.3 Audio device: Intel Corporation Device 9d70 (rev 21)
00:1f.4 SMBus: Intel Corporation Device 9d23 (rev 21)
01:00.0 Network controller: Intel Corporation Wireless 3165 (rev 79)

rd1gros@rd1gros:~/libfreenect2/build/bin$ ./Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 003559661147
[Info] [Freenect2Impl] found 1 devices
libGL error: pci id for fd 10: 8086:1916, driver (null)
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.

xlz, 2016-11-24:
You need to install this package http://packages.ubuntu.com/trusty-updates/libgl1-mesa-dri-lts-xenial

But there is some dependency of this package you need to figure out first.

iaiman, 2016-11-25:
Thank you Mr.xlz.
I'm sorry to cause you inconvenience,,, but please tell me a little bit more  about  "some dependency" ?
I'd like to get a cue to learn about this package.

xlz, 2016-12-05:
I couldn't. If you apt-get install this package, the dependencies may mess up your whole graphics stack. It's kind of hard to explain apt dependency resolution but this package is what you are looking for.

closed 2016-12-05 by xlz