OpenKinect/libfreenect2 #759: help_install

ge53, 2016-11-29:
Overview Description:
After Brew install/uninstall many times on Mac OS ElCapitan, my install keeps jamming at the "build" 
with these terminal errors:  
mkdir build && cd build
geraldirmansMBP:build geraldineerman$ cmake ..
-bash: cmake: command not found
geraldirmansMBP:build geraldineerman$ make install
make: *** No rule to make target `install'.  Stop.
I am unclear how to proceed. Any help is much appreciated
thank you.

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. 
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

xlz, 2016-11-29:
> Make sure these build tools are available: wget, git, cmake, pkg-config. Xcode may provide some of them. Install the rest via package managers.

closed 2016-11-29 by xlz

ge53, 2016-11-29:
Thank you for the prompt response--Did et all of the above, getting these
errors (below) so cannot run Protonect or NiViewer ("device not
found")--Please help...
*ld: symbol(s) not found for architecture x86_64*
*clang: error: linker command failed with exit code 1 (use -v to see
invocation)*
*make[2]: *** [lib/libfreenect2.0.2.0.dylib] Error 1*
*make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2*
*make: *** [all] Error 2*

http://myvernacular.tumblr.com

ge53, 2016-11-30:
I am still unable to install; done all the fixes you advised, tried
multiple times to install + uninstall SO many times, final error in "build"
below/again, i appreciate any advice you can offer (C++ not my thing,
really)...

ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see
invocation)
make[2]: *** [lib/libfreenect2.0.2.0.dylib] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

geraldirmansMBP:build geraldineerman$ clang -v
Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
geraldirmansMBP:build geraldineerman$

http://myvernacular.tumblr.com