OpenKinect/libfreenect2 #761: Protonect example stuck skipping rgb packet (CentOS 6)

tgpx31, 2016-11-30:
Overview Description:

Version, Platform, and Hardware Bug Found:
Version 0.2.0
Platform: CentOS 6, kernel 4.8.7

Bug found: detects Kinect2 device, starts successful. Skips finite amount of rgb packets, no viewer appears, and no other data logs.

Steps to Reproduce:
Just run ./Protonect on CentOS 6 4.8 kernel. 

Actual Results:
./Protonect
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 19 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @6:3 with serial 500012642042
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9771
[Debug] [Freenect2DeviceImpl] status 0x090000: 9771
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 500012642042
device firmware: 4.0.3911.0
[Debug] [RgbPacketStreamParser] skipping rgb packet!

It stops here and requires the process to be killed

Expected Results:
Run Protonect and display viewer.

Reproducibility:
So far 100%. Only difference is the amount of rgb packets skipped before breaking. Anywhere from 1-15.

Additional Information:
I'm not entirely familiar with Linux, this whole thing is a learning process for me and my partners. Interestingly enough, libfreenect's example for the Kinect1 works just fine. It's specifically with the Kinect2 for libfreenect2.

smokhov, 2016-12-01:
Double-check to make sure the correct libusb is linked using `ldd ./Protonect` as per #481 to see if it is not using the older system's one.

xlz, 2016-12-05:
Please report results of the following variations:
* `Protonect -noviewer`
* `Protonect -noviewer -norgb`
* `Protonect -noviewer -nodepth`

Also report `lsusb -t` and `lspci | grep -i xhci`

smokhov, 2016-12-05:
FOI #711

tgpx31, 2016-12-07:
@smokhov , ldd ./Protonect | grep libusb returns libusb-1.0.so.0 as the libusb
@xlz , here are other outputs:

**Protonect -noviewer:**
[root@localhost build]# ./Protonect -noviewer
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 18 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @6:5 with serial 500012642042
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 500012642042
device firmware: 4.0.3911.0
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!

It gets this far and stops entirely

**Protonect -noviewer -norgb:**
[root@localhost build]# ./Protonect -noviewer -norgb
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 18 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @6:5 with serial 500012642042
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 500012642042
device firmware: 4.0.3911.0

Nothing after outputting the firmware

**Protonect -noviewer -nodepth (This one seems to at least know it's getting frames):**
[root@localhost build]# ./Protonect -noviewer -nodepth
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 18 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @6:5 with serial 500012642042
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 500012642042
device firmware: 4.0.3911.0
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
The viewer is turned off. Received 300 frames. Ctrl-C to stop.
The viewer is turned off. Received 400 frames. Ctrl-C to stop.
The viewer is turned off. Received 500 frames. Ctrl-C to stop.
The viewer is turned off. Received 600 frames. Ctrl-C to stop.
The viewer is turned off. Received 700 frames. Ctrl-C to stop.

Hangs after 700 frames, requires to be killed.

I also ran ./Protonect -nodepth by itself to see output:
[root@localhost build]# ./Protonect -nodepth
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 18 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @6:5 with serial 500012642042
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 500012642042
device firmware: 4.0.3911.0
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
The viewer is turned off. Received 300 frames. Ctrl-C to stop.

xlz, 2016-12-07:
You didn't report `Also report lsusb -t and lspci | grep -i xhci`. There is nothing I can say about it.

tgpx31, 2016-12-08:
Sorry, forgot about that portion...
**lsusb -t:**
[root@localhost build]# lsusb -t
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=hub, Driver=hub/8p, 480M
        |__ Port 2: Dev 3, If 0, Class=hub, Driver=hub/3p, 480M
            |__ Port 3: Dev 6, If 0, Class=hub, Driver=hub/3p, 480M
        |__ Port 3: Dev 4, If 0, Class=HID, Driver=usbhid, 1.5M
        |__ Port 4: Dev 5, If 1, Class=HID, Driver=usbhid, 1.5M
        |__ Port 4: Dev 5, If 0, Class=HID, Driver=usbhid, 1.5M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=hub, Driver=hub/6p, 480M
        |__ Port 5: Dev 3, If 0, Class=stor., Driver=usb-storage, 480M
        |__ Port 5: Dev 3, If 1, Class=HID, Driver=usbhid, 480M

**kspci | grep - xhci:**
[root@localhost build]# lspci | grep -i xhci
[root@localhost build]# 
no output

xlz, 2016-12-08:
I can only think it's something with your USB controller. Full `lspci`? Also is there anything in dmesg related to USB?

tgpx31, 2016-12-13:
**lspci**
00:00.0 Host bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 DMI2 (rev 04)
00:01.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 1a (rev 04)
00:02.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 2a (rev 04)
00:03.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 3a (rev 04)
00:05.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 VTd/Memory Map/Misc (rev 04)
00:05.2 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 IIO RAS (rev 04)
00:05.4 PIC: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 IOAPIC (rev 04)
00:11.0 PCI bridge: Intel Corporation C600/X79 series chipset PCI Express Virtual Root Port (rev 06)
00:16.0 Communication controller: Intel Corporation C600/X79 series chipset MEI Controller #1 (rev 05)
00:19.0 Ethernet controller: Intel Corporation 82579V Gigabit Network Connection (rev 06)
00:1a.0 USB controller: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #2 (rev 06)
00:1b.0 Audio device: Intel Corporation C600/X79 series chipset High Definition Audio Controller (rev 06)
00:1c.0 PCI bridge: Intel Corporation C600/X79 series chipset PCI Express Root Port 1 (rev b6)
00:1c.1 PCI bridge: Intel Corporation C600/X79 series chipset PCI Express Root Port 2 (rev b6)
00:1c.2 PCI bridge: Intel Corporation C600/X79 series chipset PCI Express Root Port 3 (rev b6)
00:1c.3 PCI bridge: Intel Corporation C600/X79 series chipset PCI Express Root Port 4 (rev b6)
00:1c.4 PCI bridge: Intel Corporation C600/X79 series chipset PCI Express Root Port 5 (rev b6)
00:1c.5 PCI bridge: Intel Corporation C600/X79 series chipset PCI Express Root Port 6 (rev b6)
00:1c.7 PCI bridge: Intel Corporation C600/X79 series chipset PCI Express Root Port 8 (rev b6)
00:1d.0 USB controller: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #1 (rev 06)
00:1e.0 PCI bridge: Intel Corporation 82801 PCI Bridge (rev a6)
00:1f.0 ISA bridge: Intel Corporation C600/X79 series chipset LPC Controller (rev 06)
00:1f.2 SATA controller: Intel Corporation C600/X79 series chipset 6-Port SATA AHCI Controller (rev 06)
00:1f.3 SMBus: Intel Corporation C600/X79 series chipset SMBus Host Controller (rev 06)
01:00.0 VGA compatible controller: NVIDIA Corporation GK104 [GeForce GTX 760] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GK104 HDMI Audio Controller (rev a1)
06:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
07:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
08:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
09:00.0 SATA controller: ASMedia Technology Inc. ASM1062 Serial ATA Controller (rev 01)
0a:00.0 FireWire (IEEE 1394): VIA Technologies, Inc. VT6315 Series Firewire Controller (rev 01)
0b:00.0 SATA controller: Marvell Technology Group Ltd. 88SE9128 PCIe SATA 6 Gb/s RAID controller with HyperDuo (rev 11)
ff:08.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 QPI Link 0 (rev 04)
ff:09.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 QPI Link 1 (rev 04)
ff:0a.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Power Control Unit 0 (rev 04)
ff:0a.1 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Power Control Unit 1 (rev 04)
ff:0a.2 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Power Control Unit 2 (rev 04)
ff:0a.3 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Power Control Unit 3 (rev 04)
ff:0b.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 UBOX Registers (rev 04)
ff:0b.3 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 UBOX Registers (rev 04)
ff:0c.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Unicast Registers (rev 04)
ff:0c.1 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Unicast Registers (rev 04)
ff:0c.2 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Unicast Registers (rev 04)
ff:0d.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Unicast Registers (rev 04)
ff:0d.1 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Unicast Registers (rev 04)
ff:0d.2 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Unicast Registers (rev 04)
ff:0e.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Home Agent 0 (rev 04)
ff:0e.1 Performance counters: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Home Agent 0 (rev 04)
ff:0f.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Integrated Memory Controller 0 Target Address/Thermal Registers (rev 04)
ff:0f.1 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Integrated Memory Controller 0 RAS Registers (rev 04)
ff:0f.2 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Integrated Memory Controller 0 Channel Target Address Decoder Registers (rev 04)
ff:0f.3 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Integrated Memory Controller 0 Channel Target Address Decoder Registers (rev 04)
ff:0f.4 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Integrated Memory Controller 0 Channel Target Address Decoder Registers (rev 04)
[... cut ...]
ff:13.4 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 QPI Ring Registers (rev 04)
ff:13.5 Performance counters: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 QPI Ring Performance Ring Monitoring (rev 04)
ff:16.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 System Address Decoder (rev 04)
ff:16.1 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Broadcast Registers (rev 04)
ff:16.2 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Broadcast Registers (rev 04)
[root@localhost build]#

xlz, 2016-12-14:
Here is your problem. ASMedia's driver support in kernel is quite poor.
```
06:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
07:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
08:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
```

xlz, 2016-12-14:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#check-the-usb-3-controller-support-status

closed 2016-12-14 by xlz