OpenKinect/libfreenect2 #767: expose a in the libfreenect2 in the depth,rub data stream the acquisition of the from of the libusb interface

mpatalberta, 2016-12-10:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. 
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

mpatalberta, 2016-12-10:
When you are using the libfreenect2 api i do not see the actual reception time of the data from the usb interface.

mpatalberta, 2016-12-10:
You have timeStamp which is acurrate to 0.1ms. A time stamp that is expressed in real world time will allow to coordinate picture between multiple cameras across multiple machines. 
Where in the code is timeStamp set in the software?

xlz, 2016-12-10:
The timestamp of each frame is coded by the camera hardware relative to the one camera only. You can take your own wall time stamp if you want to compare multiple cameras.

closed 2016-12-10 by xlz

mpatalberta, 2016-12-10:
I am having a hard time seeing where the hardware assigns the 0.1 millisecond time stamp.
Can you please indicate where about this occurs.

xlz, 2016-12-10:
The hardware timestamp is just what is being transmitted from the hardware. We don't know where it is assigned either.