OpenKinect/libfreenect2 #769: Finding World co-ordinates

sairamkolla, 2016-12-12:
Hello there,

I'm trying to find world co-ordinates of a color pixel. For this I found the depth to color mapping using `registration->apply() ` function with the **bigdepth** parameter. Now how to find the world coordinates of any depth pixel in this range.

My understanding of `getPointXYZ()` function is that it only returns the world coordinates of depth pixel in the range 512x424. But now my depth pixels are in the range 1920x1082. 

Thankyou

xlz, 2016-12-13:
1. Get RGB camera intrinsics
2. Project a ray from a pixel in RGB image
3. Use the depth from corresponding pixel in bigdepth to find the 3D point on the ray

closed 2016-12-13 by xlz

sairamkolla, 2016-12-13:
Hello,

Can you please elaborate on the 2nd step? How to find the corresponding depth pixel for the color pixel?

I thought they have one to one correspondance i..e `colorpixel[i,j]` has the depth pixel `bigdepth[1920*i+j]`  because that's what registration from color to depth does right? Please correct me if I'm wrong.

Thanks

xlz, 2016-12-13:
`colorpixel[x,y] <-> bigdepth[x,y+1]`?

sairamkolla, 2016-12-13:
Yes, I'm asking if that holds.Thanks!

xlz, 2016-12-13:
Yes

sairamkolla, 2016-12-13:
Hi!

Actually I wanted to find the world coordinates of the corners of the chess board. So I found the bigDepth image, but the depth values of some of these corners are '0' in the bigDepth image. Can you tell why?

As @floe mentioned in #708 that bigDepth is upsampled version of depth image, how there are zeros in the bigDepth image when depth image doesn't have zeros in that ROI ( here ROI is the checkerboard)

Thanks

**Depth Image**
[image]

**bigDepth image**
[image]

**bigDepth values of some corners**
[image]

xlz, 2016-12-13:
> here are zeros in the bigDepth image when depth image doesn't have zeros in that ROI 
I don't see any of the case.

sairamkolla, 2016-12-13:
@xlz  in the 3rd image, as you can see the last column is the bigDepth value of that particular corner. Most of them are non zeros and are almost equal(since the board is in vertical position). But some values are zeros.

The code I used to access the bigDepth value is 

`depth_value  = float(bigDepth.data[int(corners[i].x),int(corners[i].y) + 1])`

Thankyou

xlz, 2016-12-13:
The small depth image also has zeros in that ROI, so what's the problem?

sairamkolla, 2016-12-13:
My ROI is the chessboard only. In the small depth image there are no zeros in that region. But I get zeros from the bigDepth image.

xlz, 2016-12-13:
I think these are zeros
[image]

sairamkolla, 2016-12-13:
ok, I still don't understand why I'm getting the checkerboard pattern in the depth image. Any idea? Sometimes I get complete grayscale image and sometimes this pattern(Hence the zeros in this case).

Also,is there any other approach for finding the world coordinates?

PS: I'm trying to implement icp algorithm for finding R and T matrices for calibration. For this I need world coordinates of the corner points.

Thankyou

xlz, 2016-12-13:
I think the zero areas are dark blocks with no reflection. ToF cameras need reflection to get depth signals. An alternative would be fitting a surface with the checkerboard so you have all points on the surface.

sairamkolla, 2016-12-14:
>  fitting a surface with the checkerboard

Can you elaborate what you mean by this? Currently my checkerboard is on a metal sheet. 

Thanks for your reply.

xlz, 2016-12-14:
http://pointclouds.org/documentation/tutorials/planar_segmentation.php#planar-segmentation