OpenKinect/libfreenect2 #770: Submitting depth transfers fails

asteinh, 2016-12-13:
This might be more related to libusb than to libfreekinect2, but maybe there's someone who encountered similar issues or has an idea how to resolve this. *Any hints are very welcome!*

**The situation**
We want to use libfreekinect2 on a research PC running Xenomai 2.6.4 - and this is also the biggest limitation: Since we rely on other software (working with this kernel version only), we cannot update or change the kernel.

**The problem**
No issues during installation, but when we run the protonect example it fails - see the output below. Using the `-nodepth` flag succeeds and the RGB image is shown.

Things that did not solve this issue:
- Various usbfs_memory_mb settings (16, 128, 512, 1024, ...)
- System's libusb (1.20 @ `/lib64/libusb-1.0.so`)
- Various values of `MAX_ISO_BUFFER_LENGTH` in libusb's `linux_usbfs.h` (49152 * 128, 49152, ...)

Tracking this down, I only got as far as: The `ioctl()` call when submitting URBs in `linux_usbfs.c` (line 2039) fails with `errno = EINVAL` and `submit_iso_transfer` therefore returns `LIBUSB_ERROR_INVALID_PARAM`.

**Some details**
Kernel versions:
```
$ uname -r
3.8.13-xenomai-2.6.4-rtnet-7c8ba1-20151020.x86_64
```

The usbfs_memory_mb setting (set via usbcore parameter):
```
$ cat /sys/module/usbcore/parameters/usbfs_memory_mb
1024
```

The linked usblib of Protonect:
```
$ ldd Protonect | grep usb
libusb-1.0.so.0 => /home/user/software/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f7f321ef000)
```

<details><summary>USB controller</summary><pre><code>
$ lspci | grep USB
00:14.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller (rev 04)
00:1a.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1d.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #1 (rev 04)
</code></pre></details>
<br />
<details><summary>USB busses</summary><pre><code>
$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 1: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
</code></pre></details>
<br />
<details><summary>dmesg when (re)plugging the Kinect</summary><pre><code>
[ 1056.132935] xhci_hcd 0000:00:14.0: Timeout while waiting for reset device command
[ 1056.132951] usb 3-2: reset high-speed USB device number 2 using xhci_hcd
[ 1056.145068] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8803d4a9ec40
[ 1056.145072] xhci_hcd 0000:00:14.0: Trying to add endpoint 0x81 without dropping it.
[ 1056.145075] usb 3-2: Busted HC?  Not enough HCD resources for old configuration.
[ 1056.145088] usb 3-2: USB disconnect, device number 2
[ 1056.145152] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8803d4a9ec40
[ 1056.145231] xhci_hcd 0000:00:14.0: Slot 1 endpoint 2 not removed from BW list!
[ 1056.401148] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[ 1056.414187] usb 3-2: New USB device found, idVendor=045e, idProduct=02d9
[... 221 lines cut ...]
libusb: warning [submit_iso_transfer] submiturb failed, transfer too large
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_INVALID_PARAM Invalid parameter
libusb: warning [submit_iso_transfer] submiturb failed, transfer too large
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_INVALID_PARAM Invalid parameter
[Error] [usb::TransferPool] all submissions failed. Try debugging with environment variable: LIBUSB_DEBUG=3.
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
Segmentation fault (core dumped)
</code></pre></details>
<br />
<details><summary>dmesg of this run</summary><pre><code>
[ 1538.014324] usb 4-2.1: reset SuperSpeed USB device number 5 using xhci_hcd
[ 1538.025532] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8803d8574200
[ 1538.025535] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8803d8574240
[ 1538.025538] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8803d85742c0
[ 1538.025540] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8803d8574280
[ 1538.086746] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 6 ep 2 with no TDs queued?
[ 1538.370483] usb 4-2.1: usbfs: process 6268 (USB) did not claim interface 0 before use
[ 1538.371364] USB[6268]: segfault at 1cf937c8 ip 00007fd97f88950a sp 00007fd97871ac60 error 4 in libusb-1.0.so.0.1.0[7fd97f87b000+17000]
</code></pre></details>
<br />

Note: the entry `usb 4-2.1: usbfs: process ... (USB) did not claim interface 0 before use` is not always shown - for some failing runs, this line does not appear in the dmesg output.

xlz, 2016-12-13:
USB 3.0 packet length limit is raised in kernel 3.10. The raised limit is required for USB 3. See this commit https://github.com/torvalds/linux/commit/e2e2f0ea1c935edcf53feb4c4c8fdb4f86d57dd9

Since it's a simple commit maybe you can patch xenomai yourself and see how it goes.

xlz, 2016-12-13:
I mean it may even be possible to directly patch the kernel binary since it's probably just a change of 4 bytes.

asteinh, 2016-12-17:
Thanks for this hint - that's exactly what I had to do!

asteinh, 2016-12-19:
I'm afraid I have to reopen this...
I changed the iso-package-length in devio.c to 49152 and recompiled the kernel, now the USB communication seems to work (at least, I don't get any error messages), but the viewer window stays black. As before, `-nodepth` works fine (showing the RGB image in the viewer), but as soon as I try to transmit IR/depth information, it fails. The viewer window is unresponsive and I have to kill the whole process.
I put the `dmesg` output and an excerpt of the debug output of `LIBUSB_DEBUG=4 ./Protonect cpu` in [this gist](https://gist.github.com/asteinh/deb4522b08a763214490b2a831c0820e).

The only strange thing that I found so far are [lines 63-68](https://gist.github.com/asteinh/deb4522b08a763214490b2a831c0820e#file-dmesg-L64) in the dmesg output - I'm not sure how to interpret this though...

xlz, 2016-12-19:
Does `Protonect -noviewer`  print framerate?

xlz, 2016-12-19:
Yeah, this line is the problem
```
[  381.487175] xhci_hcd 0000:00:14.0: xHCI host not responding to stop endpoint command.
[  381.487178] xhci_hcd 0000:00:14.0: Assuming host is dying, halting host.
[  381.487218] xhci_hcd 0000:00:14.0: HC died; cleaning up
```
I think this is simply your kernel not able to support the USB 3 hardware.

asteinh, 2016-12-19:
`Protonect -noviewer` prints the framerate, yes.

The mentioned lines only appear *after* killing the process - I thought that probably causes these errors?

xlz, 2016-12-19:
If it can print framerate you may get the data without the viewer.

asteinh, 2016-12-19:
I was wrong, the framecounter is not increased and the framerate is not printed.

As far as I could track it down - for now - it seems that the bits for IR/depth frames are never set to "ready", so [this line](https://github.com/OpenKinect/libfreenect2/blob/83f88b4c09f0b00724ae65785abcd4f3eeb79f93/src/frame_listener_impl.cpp#L186) is only reached once for the RGB image and `ready_frame_types_` is either 0 or 1, but never 7. I'd guess that the call of `onNewFrame` for a depth package should come from [this function](https://github.com/OpenKinect/libfreenect2/blob/83f88b4c09f0b00724ae65785abcd4f3eeb79f93/src/cpu_depth_packet_processor.cpp#L879I), but although I use the `cpu` flag, it is never called.

xlz, 2016-12-19:
I don't have anything other than https://github.com/OpenKinect/libfreenect2/issues/770#issuecomment-267950402

asteinh, 2016-12-19:
Ok, thanks for your effort! I'll look for solutions w.r.t. the kernel then.

closed 2016-12-19 by asteinh