OpenKinect/libfreenect2 #771: Question: Good Single Board Computer for Kinect 2

k15a, 2016-12-14:
Hey,

Does anyone has some experience or recommendation which single board computers work best with the Kinect 2?

boulabiar, 2016-12-14:
I haven't tested any option, but I share an information that may be useful.
Intel is releasing a board that has USB3.0 which is needed for Kinect2:
http://click.intel.com/intelr-realsensetm-robotic-development-kit.html
(maybe it is the UP board ?)

Other people here used Intel NUC
http://www.intel.com/content/www/us/en/nuc/overview.html

i

hanshammel1337, 2016-12-21:
You can also have a look at Nvidias mobile computing platforms Jetson TK1 or TX1

https://www.nvidia.com/object/jetson-tk1-embedded-dev-kit.html
https://www.nvidia.com/object/jetson-tx1-dev-kit.html

boulabiar, 2017-01-05:
On Thu, Jan 5, 2017 at 10:11 AM, Istvan Rath <notifications@github.com>
wrote:

> http://www.lattepanda.com/
>
> Looks very interesting as a board.
Have you tested it with libfreenect2 ?
I don't know whether the 2GB is sufficient.

i

istvanrath, 2017-01-06:
Me personally not, but there are remarks on the forum (e.g. http://www.lattepanda.com/forum/viewtopic.php?f=6&t=848&p=3652&hilit=kinect#p3652) that it works, although very slowly. Which is to be expected, given the low specs of the board with respect to the recommended configuration by Microsoft (https://developer.microsoft.com/en-us/windows/kinect/hardware-setup).

So I guess if performance is critical in your application you need to go with something more powerful, i.e. something like https://www.digikey.com/product-detail/en/advantech-corp/MIO-5290U-S6A1E/MIO-5290U-S6A1E-ND/5272444

philipNoonan, 2017-03-14:
I've used a lattepanda with the v2 MS SDK and I was pretty surprised to see it work OK with fusion. Nothing great, 10-20 fps at low(ish) resolutions, but still, I was fairly impressed.