OpenKinect/libfreenect2 #772: Accessing values in color depth map

navyasri1, 2016-12-14:
Hello,
According to [Documentation](https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Registration.html#a8be3a548ff440e30f5e89a4b053e04fa) `color_depth_map` gives the index of mapped color pixel.But what I see here is it has negative(not only -1) values and also zeros in it.Any idea why this occurs. 

Thanks

navyasri1, 2016-12-14:
I am accessing data in this 
`value = color_to_depth[512*xi+yi]` xi and yi are coordinates.
So i should have value/1920 and value%1920 as the indices of mapped color pixel ideally.But here I am getting less number of values which are greater than zero(these values have high magnitudes) and many values are zero(this should not be the case as zero indicates one index and depth values does not map to same color pixel) and some values are even negative.Correct me if I am wrong.

What do you mean by `what does the data look like`?

xlz, 2016-12-14:
What does `bigdepth` look like? If there is no color correspondence for a depth pixel `color_depth_map` won't have the correspondence.

closed 2017-01-20 by navyasri1

sairamkolla, 2017-02-21:
Hi,

Below are the bigepth and my rgb image.
[image]
[image]

I have iterated over all the depth pixels and colored their corresponding rgb pixels. But I get a very wierd pattern as shown.

Any idea why this is happening? 

The code 
```
int **color_to_depth[512*424];
registration->apply(rgb, depth, &undistorted, &registered,true,&bigDepth,color_to_depth);

for (int xi=0; xi<424; ++xi){
            for (int yi=0; yi<512; ++yi){
                registration->getPointXYZ(&undistorted, xi,yi, X,Y,Z);
                if(X!=X || Y!=Y || Z!=Z){} //Ignore the black areas
                else if(color_to_depth[512*xi+yi] >0){
                        int index = color_to_depth[512*xi+yi];
                        circle(temprgbMat, cvPoint(index/1920,index%1920), 1, cv::Scalar(0.0, 0.0, 255.0, 0.0),-7, 8, 0); 
                }
           }
}
```

floe, 2017-02-22:
AFAICT you don't need `color_to_depth` here at all, if you already have `bigdepth`. Just iterate over the bigdepth frame and set the corresponding pixel in the RGB image.

sairamkolla, 2017-02-22:
@floe I've thought of doing that. But what is the formulation for getting world coordinates from these x,y,z ?
(x,y from rgb and z from bigdepth)

sairamkolla, 2017-02-22:
[image]

On the other hand when I'm coloring (index/1080,index%1080) I'm getting this image. Which one is correct?

This image is taken in a different pose. But the pattern I get is similar.

JimTheKiwi, 2017-02-22:
The first image example looks reflected around a 45 degree line from the top left, like x and y are swapped in the red overlay, or the array storage order is confusing row-major vs column-major.  Maybe test (index%1920,index/1920)

sairamkolla, 2017-02-25:
@JimTheKiwi Thanks! It worked.