OpenKinect/libfreenect2 #773: Build Error on Linux Mint 17.3 Rosa

byronprice, 2016-12-21:
Trying to build and install libfreenect2 with Linux Mint 17.3 Rosa.

The command, 'cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2' seems to work fine.
Get the following Feature list:
-- Feature list:
--   CUDA    no
--   CXX11    disabled
--   Examples    yes
--   OpenCL    no
--   OpenGL    yes
--   OpenNI2    no
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jglab/libfreenect2/build

Next command, 'make' sends an error:

[ 60%] Building CXX object CMakeFiles/freenect2.dir/src/command_transaction.cpp.o
/home/jglab/libfreenect2/src/command_transaction.cpp: In member function ‘bool libfreenect2::protocol::CommandTransaction::isResponseCompleteResult(libfreenect2::protocol::CommandTransaction::Result&, uint32_t)’:
/home/jglab/libfreenect2/src/command_transaction.cpp:127:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
[ 64%] Building CXX object CMakeFiles/freenect2.dir/src/registration.cpp.o
[ 68%] Building CXX object CMakeFiles/freenect2.dir/src/logging.cpp.o
[ 72%] Building CXX object CMakeFiles/freenect2.dir/src/libfreenect2.cpp.o
[ 76%] Building CXX object CMakeFiles/freenect2.dir/src/tinythread/tinythread.cpp.o
[ 80%] Building CXX object CMakeFiles/freenect2.dir/src/turbo_jpeg_rgb_packet_processor.cpp.o
[ 84%] Building CXX object CMakeFiles/freenect2.dir/src/flextGL.cpp.o
[ 88%] Building CXX object CMakeFiles/freenect2.dir/src/opengl_depth_packet_processor.cpp.o
Linking CXX shared library lib/libfreenect2.so
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/4.7/../../../x86_64-linux-gnu/libturbojpeg.a(libturbojpeg_la-turbojpeg.o): relocation R_X86_64_32 against `.data' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/x86_64-linux-gnu/4.7/../../../x86_64-linux-gnu/libturbojpeg.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lib/libfreenect2.so.0.2.0] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

xlz, 2016-12-21:
`make VERBOSE=1`?

I'm not sure why it's linking the static `libturbojpeg.a`.

byronprice, 2016-12-21:
~/libfreenect2/build $ make VERBOSE=1
/usr/bin/cmake -H/home/jglab/libfreenect2 -B/home/jglab/libfreenect2/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/jglab/libfreenect2/build/CMakeFiles /home/jglab/libfreenect2/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory `/home/jglab/libfreenect2/build'
make -f CMakeFiles/generate_resources_tool.dir/build.make CMakeFiles/generate_resources_tool.dir/depend
make[2]: Entering directory `/home/jglab/libfreenect2/build'
cd /home/jglab/libfreenect2/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/jglab/libfreenect2 /home/jglab/libfreenect2 /home/jglab/libfreenect2/build /home/jglab/libfreenect2/build /home/jglab/libfreenect2/build/CMakeFiles/generate_resources_tool.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/jglab/libfreenect2/build'
make -f CMakeFiles/generate_resources_tool.dir/build.make CMakeFiles/generate_resources_tool.dir/build
make[2]: Entering directory `/home/jglab/libfreenect2/build'
make[2]: Nothing to be done for `CMakeFiles/generate_resources_tool.dir/build'.
make[2]: Leaving directory `/home/jglab/libfreenect2/build'
/usr/bin/cmake -E cmake_progress_report /home/jglab/libfreenect2/build/CMakeFiles  25
[  4%] Built target generate_resources_tool
make -f CMakeFiles/freenect2.dir/build.make CMakeFiles/freenect2.dir/depend
make[2]: Entering directory `/home/jglab/libfreenect2/build'
cd /home/jglab/libfreenect2/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/jglab/libfreenect2 /home/jglab/libfreenect2 /home/jglab/libfreenect2/build /home/jglab/libfreenect2/build /home/jglab/libfreenect2/build/CMakeFiles/freenect2.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/jglab/libfreenect2/build'
make -f CMakeFiles/freenect2.dir/build.make CMakeFiles/freenect2.dir/build
make[2]: Entering directory `/home/jglab/libfreenect2/build'
Linking CXX shared library lib/libfreenect2.so
/usr/bin/cmake -E cmake_link_script CMakeFiles/freenect2.dir/link.txt --verbose=1
/usr/bin/c++  -fPIC  -Wall -O2 -g -DNDEBUG   -shared -Wl,-soname,libfreenect2.so.0.2 -o lib/libfreenect2.so.0.2.0 CMakeFiles/freenect2.dir/src/transfer_pool.cpp.o CMakeFiles/freenect2.dir/src/event_loop.cpp.o CMakeFiles/freenect2.dir/src/usb_control.cpp.o CMakeFiles/freenect2.dir/src/allocator.cpp.o CMakeFiles/freenect2.dir/src/frame_listener_impl.cpp.o CMakeFiles/freenect2.dir/src/packet_pipeline.cpp.o CMakeFiles/freenect2.dir/src/rgb_packet_stream_parser.cpp.o CMakeFiles/freenect2.dir/src/rgb_packet_processor.cpp.o CMakeFiles/freenect2.dir/src/depth_packet_stream_parser.cpp.o CMakeFiles/freenect2.dir/src/depth_packet_processor.cpp.o CMakeFiles/freenect2.dir/src/cpu_depth_packet_processor.cpp.o CMakeFiles/freenect2.dir/src/resource.cpp.o CMakeFiles/freenect2.dir/src/command_transaction.cpp.o CMakeFiles/freenect2.dir/src/registration.cpp.o CMakeFiles/freenect2.dir/src/logging.cpp.o CMakeFiles/freenect2.dir/src/libfreenect2.cpp.o CMakeFiles/freenect2.dir/src/tinythread/tinythread.cpp.o CMakeFiles/freenect2.dir/src/turbo_jpeg_rgb_packet_processor.cpp.o CMakeFiles/freenect2.dir/src/flextGL.cpp.o CMakeFiles/freenect2.dir/src/opengl_depth_packet_processor.cpp.o -lusb-1.0 -lpthread -Wl,-Bstatic -lturbojpeg -Wl,-Bdynamic -lglfw -lGL 
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/4.7/../../../x86_64-linux-gnu/libturbojpeg.a(libturbojpeg_la-turbojpeg.o): relocation R_X86_64_32 against `.data' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/x86_64-linux-gnu/4.7/../../../x86_64-linux-gnu/libturbojpeg.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lib/libfreenect2.so.0.2.0] Error 1
make[2]: Leaving directory `/home/jglab/libfreenect2/build'
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make[1]: Leaving directory `/home/jglab/libfreenect2/build'
make: *** [all] Error 2

xlz, 2016-12-21:
I don't know how to search for packages with Mint. This http://packages.linuxmint.com/search.php?release=any&section=any&keyword=jpeg gives me zero results.

Anyway it looks like CMake didn't find the dynamic library of  turbojpeg. You have to tell me how is turbojpeg packaged on Linux Mint and the file list of the turbojpeg package.

andre-nguyen, 2017-02-10:
I'm getting the same thing on ubuntu 16, any ideas?

Ubuntu version
```
andre@neptune:~/libfreenect2/build$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial
```

Cmake output
```
andre@neptune:~/libfreenect2/build$ cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 
-- using tinythread as threading library
-- Checking for modules 'libva;libva-drm'
--   No package 'libva' found
--   No package 'libva-drm' found
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Could NOT find OpenCL (missing:  OpenCL_LIBRARY OpenCL_INCLUDE_DIR) 
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 pthread
 /usr/lib/x86_64-linux-gnu/libturbojpeg.a
 /usr/lib/x86_64-linux-gnu/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing:  OpenNI2_LIBRARY_DIR OpenNI2_INCLUDE_DIRS) 
-- Feature list:
--   CUDA    no
--   CXX11    disabled
--   Examples    yes
--   OpenCL    no
--   OpenGL    yes
--   OpenNI2    no
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/andre/libfreenect2/build
```

make output
```
andre@neptune:~/libfreenect2/build$ make VERBOSE=1
/usr/bin/cmake -H/home/andre/libfreenect2 -B/home/andre/libfreenect2/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/andre/libfreenect2/build/CMakeFiles /home/andre/libfreenect2/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/andre/libfreenect2/build'
make -f CMakeFiles/generate_resources_tool.dir/build.make CMakeFiles/generate_resources_tool.dir/depend
make[2]: Entering directory '/home/andre/libfreenect2/build'
cd /home/andre/libfreenect2/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/andre/libfreenect2 /home/andre/libfreenect2 /home/andre/libfreenect2/build /home/andre/libfreenect2/build /home/andre/libfreenect2/build/CMakeFiles/generate_resources_tool.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/andre/libfreenect2/build'
make -f CMakeFiles/generate_resources_tool.dir/build.make CMakeFiles/generate_resources_tool.dir/build
make[2]: Entering directory '/home/andre/libfreenect2/build'
make[2]: Nothing to be done for 'CMakeFiles/generate_resources_tool.dir/build'.
make[2]: Leaving directory '/home/andre/libfreenect2/build'
[  7%] Built target generate_resources_tool
make -f CMakeFiles/freenect2.dir/build.make CMakeFiles/freenect2.dir/depend
make[2]: Entering directory '/home/andre/libfreenect2/build'
cd /home/andre/libfreenect2/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/andre/libfreenect2 /home/andre/libfreenect2 /home/andre/libfreenect2/build /home/andre/libfreenect2/build /home/andre/libfreenect2/build/CMakeFiles/freenect2.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/andre/libfreenect2/build'
make -f CMakeFiles/freenect2.dir/build.make CMakeFiles/freenect2.dir/build
make[2]: Entering directory '/home/andre/libfreenect2/build'
[ 10%] Linking CXX shared library lib/libfreenect2.so
/usr/bin/cmake -E cmake_link_script CMakeFiles/freenect2.dir/link.txt --verbose=1
/usr/bin/c++  -fPIC  -Wall -O2 -g -DNDEBUG  -shared -Wl,-soname,libfreenect2.so.0.2 -o lib/libfreenect2.so.0.2.0 CMakeFiles/freenect2.dir/src/transfer_pool.cpp.o CMakeFiles/freenect2.dir/src/event_loop.cpp.o CMakeFiles/freenect2.dir/src/usb_control.cpp.o CMakeFiles/freenect2.dir/src/allocator.cpp.o CMakeFiles/freenect2.dir/src/frame_listener_impl.cpp.o CMakeFiles/freenect2.dir/src/packet_pipeline.cpp.o CMakeFiles/freenect2.dir/src/rgb_packet_stream_parser.cpp.o CMakeFiles/freenect2.dir/src/rgb_packet_processor.cpp.o CMakeFiles/freenect2.dir/src/depth_packet_stream_parser.cpp.o CMakeFiles/freenect2.dir/src/depth_packet_processor.cpp.o CMakeFiles/freenect2.dir/src/cpu_depth_packet_processor.cpp.o CMakeFiles/freenect2.dir/src/resource.cpp.o CMakeFiles/freenect2.dir/src/command_transaction.cpp.o CMakeFiles/freenect2.dir/src/registration.cpp.o CMakeFiles/freenect2.dir/src/logging.cpp.o CMakeFiles/freenect2.dir/src/libfreenect2.cpp.o CMakeFiles/freenect2.dir/src/tinythread/tinythread.cpp.o CMakeFiles/freenect2.dir/src/turbo_jpeg_rgb_packet_processor.cpp.o CMakeFiles/freenect2.dir/src/flextGL.cpp.o CMakeFiles/freenect2.dir/src/opengl_depth_packet_processor.cpp.o -lusb-1.0 -lpthread -Wl,-Bstatic -lturbojpeg -Wl,-Bdynamic -lglfw -lGL 
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libturbojpeg.a(libturbojpeg_la-turbojpeg.o): relocation R_X86_64_32 against `.data' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libturbojpeg.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
CMakeFiles/freenect2.dir/build.make:602: recipe for target 'lib/libfreenect2.so.0.2.0' failed
make[2]: *** [lib/libfreenect2.so.0.2.0] Error 1
make[2]: Leaving directory '/home/andre/libfreenect2/build'
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/freenect2.dir/all' failed
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make[1]: Leaving directory '/home/andre/libfreenect2/build'
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

dg-w36, 2017-03-19:
Hi @andre-nguyen, I got the same error with you. 
I run 
`sudo apt-get autoremove libturbojpeg libjpeg-turbo8-dev `
and run 
`sudo apt-get install libturbojpeg libjpeg-turbo8-dev `
to install them again.  And then run make successfully.

xlz, 2017-03-28:
Can't reproduce.

I got a Mint 17.3 live cd and built fine.

closed 2017-03-28 by xlz