OpenKinect/libfreenect2 #775: some error appeared during complete with catkin_make in ROS

XeMinZa, 2016-12-23:
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clReleaseMemObject@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clReleaseCommandQueue@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clEnqueueWriteBuffer@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clReleaseDevice@OPENCL_1.2'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clSetKernelArg@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clRetainKernel@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clCreateContext@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clReleaseProgram@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clGetPlatformInfo@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clEnqueueUnmapMemObject@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clCreateProgramWithSource@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clCreateKernel@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clRetainContext@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clCreateCommandQueue@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clWaitForEvents@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clEnqueueReadBuffer@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clEnqueueNDRangeKernel@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clEnqueueMapBuffer@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clRetainProgram@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clRetainMemObject@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clGetProgramBuildInfo@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clRetainEvent@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clReleaseKernel@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clReleaseEvent@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clCreateBuffer@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clBuildProgram@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clGetDeviceIDs@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clRetainDevice@OPENCL_1.2'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clRetainCommandQueue@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clGetDeviceInfo@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clReleaseContext@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clGetExtensionFunctionAddress@OPENCL_1.0'
/home/robot508/freenect2/lib/libfreenect2.so: undefined reference to `clGetPlatformIDs@OPENCL_1.0'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/robot508/catkin_ws/devel/lib/get_ball_location/get_ball_location] Error 1
make[1]: *** [get_ball_location/CMakeFiles/get_ball_location.dir/all] Error 2
make: *** [all] Error 2

# I can run the example,but when I complete it with catkiin_make in ROS .error appeared

XeMinZa, 2016-12-26:
I see how to aviod the error. Just don't select the OpenCl option when compiling the libiary...And the error will disappear. While it don't know why it work,it did.

xlz, 2017-03-14:
This issue is now a duplicate of #804.

closed 2017-03-14 by xlz

liuhang0727, 2018-08-18:
I am troubled in the same problem, I have tryed the solution in #804, but it still can't work. 
Could you give me some advice? Or teach me how to turn off the OpenCl option when compiling the libiary.
Thank you.