OpenKinect/libfreenect2 #777: Missing helper_math.h

gunnicus, 2016-12-28:
I'm trying to build libfreenect2 x64 on Windows 10 with Visual Studio 2013

But the build failed with error: Cannot open include file: 'helper_math.h': No such file or directory 

Build FAILED.

"C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\install.vcxproj" (default target) (1) ->
"C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\ALL_BUILD.vcxproj" (default target) (3) ->
"C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\examples\Protonect.vcxproj" (default target) (4) ->
"C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\freenect2.vcxproj" (default target) (5) ->
(CustomBuild target) ->
  C:/Users/TODD/Documents/Kinect/libfreenect2-master/src/cuda_depth_packet_processor.cu(31): fatal error C1083: Cannot open include file: 'helper_math.h': No such file or directory [C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\freenect2.vcxproj]

    0 Warning(s)
    1 Error(s)

I was following the windows installation steps:

-- Feature list:
--   CUDA    yes
--   CXX11    disabled
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/TODD/Documents/Kinect/libfreenect2-master/build

gaborpapp, 2017-01-19:
`helper_math.h` should be in the CUDA samples directory. `/Developer/NVIDIA/CUDA-8.0/samples/common/inc/helper_math.h` on OSX. Install the samples or try to build without CUDA.

RajaMOeez, 2017-02-18:
Hello Sir Im also facing the same problem @gunnicus  please let me if you find the problem solution or if not vice versa I'll help you if you didn't find.
[image]

xlz, 2017-02-20:
`NVCUDASAMPLES8_0_ROOT` or `NVCUDASAMPLES_ROOT ` should be among your environment variables. Are they?

```
C:\Users\stars\Documents\GitHub\libfreenect2\build [master]> ls $env:NVCUDASAMPLES8_0_ROOT/common/inc/helper_math.h

    Directory: C:\ProgramData\NVIDIA Corporation\CUDA Samples\v8.0\common\inc

Mode                LastWriteTime     Length Name
----                -------------     ------ ----
-a---        10/28/2015   5:07 AM      38763 helper_math.h
```

xlz, 2017-03-29:
Can't reproduce.

closed 2017-03-29 by xlz

rickyzhang82, 2017-12-07:
I can reproduce this in Fedora 4.13.13-300.fc27.x86_64.

You should never assume that cuda user install sample. Neither should you assume that env variable is set.

Can we make this independent from header file from cuda sample? No offense. But that sounds a little bit silly from me.

xlz, 2017-12-07:
We don't assume cuda users install sample. We require.

The sample header is copyright 1993-2012 NVIDIA Corporation all rights reserved. We cannot redistribute.

I'll add some documentation about this though.

Fighter777, 2018-03-06:
hello, how to disable cuda ? i have the same problem...

MohameHassan, 2018-12-10:
You disable cuda by running
`cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 -ENABLE_CUDA OFF`

sohumango, 2019-04-04:
1. install Cuda toolkits
   sudo ./cuda_10.1.105_418.39_linux.run
2. find  helper_math.h, and export it to CPATH
  /home/xxxx/NVIDIA_CUDA-10.1_Samples/common/inc/helper_math.h
  export CPATH=$CPATH:$HOME/NVIDIA_CUDA-10.1_Samples/common/inc
3. make after cmake.

XYudong, 2019-09-03:
> You can disable cuda by running
> `cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 -ENABLE_CUDA OFF`

probably `cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 -DENABLE_CUDA=OFF`

remmel, 2021-12-20:
Alternative to https://github.com/OpenKinect/libfreenect2/issues/777#issuecomment-480023146 
the cuda toolkit can be install directly from apt if your card is not too old:
`sudo apt install nvidia-cuda-toolkit`
and later to get the header:
`git clone git@github.com:NVIDIA/cuda-samples.git` (select the tag matching your cuda toolkit version)
`export CPATH=$CPATH:/home/xxx/workspace/cuda-samples/Common/`

WyattAutomation, 2022-10-28:
> Alternative to [#777 (comment)](https://github.com/OpenKinect/libfreenect2/issues/777#issuecomment-480023146) the cuda toolkit can be install directly from apt if your card is not too old: `sudo apt install nvidia-cuda-toolkit` and later to get the header: `git clone git@github.com:NVIDIA/cuda-samples.git` (select the tag matching your cuda toolkit version) `export CPATH=$CPATH:/home/xxx/workspace/cuda-samples/Common/`

This resolves it for me, Ubuntu 22.04