OpenKinect/libfreenect2 #778: freenect2-openni2 build error : error LINK 1169, error LINK2005

gunnicus, 2016-12-28:
I'm using visual studio 2013 to build libfreenect2 solution. x64 Windows 10 kinect2.

But there are building errors in project freenect2-openni2

Error	8	error LNK1169: one or more multiply defined symbols found	C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\bin\Debug\freenect2-openni2d.dll	1	1	freenect2-openni2
Error	1	error LNK2005: "public: static int const Freenect2Driver::DepthStream::MAX_VALUE" (?MAX_VALUE@DepthStream@Freenect2Driver@@2HB) already defined in DepthStream.obj	C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\DeviceDriver.obj	freenect2-openni2
Error	3	error LNK2005: "public: static unsigned __int64 const Freenect2Driver::DepthStream::CONST_SHIFT_VAL" (?CONST_SHIFT_VAL@DepthStream@Freenect2Driver@@2_KB) already defined in DepthStream.obj	C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\DeviceDriver.obj	freenect2-openni2
Error	2	error LNK2005: "public: static unsigned __int64 const Freenect2Driver::DepthStream::GAIN_VAL" (?GAIN_VAL@DepthStream@Freenect2Driver@@2_KB) already defined in DepthStream.obj	C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\DeviceDriver.obj	freenect2-openni2
Error	4	error LNK2005: "public: static unsigned __int64 const Freenect2Driver::DepthStream::MAX_SHIFT_VAL" (?MAX_SHIFT_VAL@DepthStream@Freenect2Driver@@2_KB) already defined in DepthStream.obj	C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\DeviceDriver.obj	freenect2-openni2
Error	5	error LNK2005: "public: static unsigned __int64 const Freenect2Driver::DepthStream::PARAM_COEFF_VAL" (?PARAM_COEFF_VAL@DepthStream@Freenect2Driver@@2_KB) already defined in DepthStream.obj	C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\DeviceDriver.obj	freenect2-openni2
Error	6	error LNK2005: "public: static unsigned __int64 const Freenect2Driver::DepthStream::SHIFT_SCALE_VAL" (?SHIFT_SCALE_VAL@DepthStream@Freenect2Driver@@2_KB) already defined in DepthStream.obj	C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\DeviceDriver.obj	freenect2-openni2
Error	7	error LNK2005: "public: static unsigned __int64 const Freenect2Driver::DepthStream::ZERO_PLANE_DISTANCE_VAL" (?ZERO_PLANE_DISTANCE_VAL@DepthStream@Freenect2Driver@@2_KB) already defined in DepthStream.obj	C:\Users\TODD\Documents\Kinect\libfreenect2-master\build\DeviceDriver.obj	freenect2-openni2

SirDifferential, 2017-01-12:
I saw this problem with MSVC 2010 64-bit, while MSVC 2013 64-bit compiled the OpenNI2 driver with no errors. I didn't look much into this as MSVC 2010 isn't officially supported according to the instructions, so I just quickly changed those constants into preprocessor defines that allowed me to get a build done.

xlz, 2017-02-20:
I don't know? People are getting weird errors on Windows which I've never been able to reproduce. This particular one neither.

xlz, 2017-03-29:
Can't reproduce.

closed 2017-03-29 by xlz

AlexMorgand, 2018-01-04:
Had the same issue (VS2013). 

Commenting the static variables in DepthStream.cpp fixed the issue for me (and those constants seem unused in the code anyway).