OpenKinect/libfreenect2 #780: Zero Intensity value for a valid X,Y,Z coordinate

mpatalberta, 2017-01-13:
Overview Description:

When getting the intensity value  I have noticed the the value is zero for a valid x,y,z depth value
Version, Platform, and Hardware Bug Found:

xlz, 2017-01-13:
https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Frame.html#a1d1cfd8ffb84e947f82999c682b666a7

> Depth 	512x424 float, unit: millimeter. Non-positive, NaN, and infinity are invalid or missing data.

mpatalberta, 2017-01-13:
In the code below I have noticed that the use of atan2 without any care for the case when both a and b terms are near zero.

/*******************************************************************************
 * Process pixel stage 2
 ******************************************************************************/
void kernel processPixelStage2(global const float3 *a_in, global const float3 *b_in, global const float *x_table, global const float *z_table,
                               global float *depth, global float *ir_sums)
{
  const uint i = get_global_id(0);
  float3 a = a_in[i];
  float3 b = b_in[i];

  float3 phase = atan2(b, a);
  phase = select(phase, phase + 2.0f * M_PI_F, isless(phase, (float3)(0.0f)));
  phase = select(phase, (float3)(0.0f), isnan(phase));
  float3 ir = sqrt(a * a + b * b) * AB_MULTIPLIER;

  float ir_sum = ir.x + ir.y + ir.z;
  float ir_min = min(ir.x, min(ir.y, ir.z));
  float ir_max = max(ir.x, max(ir.y, ir.z));

mpatalberta, 2017-01-13:
When getting the intensity value I have noticed the the value is occasionally zero for a valid x,y,z depth value. Is this an expected behavior and what does this mean?

xlz, 2017-01-14:
> what does this mean?

noise?

mpatalberta, 2017-01-14:
Not certain.

I have identified a potential error in the software when using OpenCL. The atan2 function is undefined for the case of both numerator and denominator are near zero or small. The value of atan2 is undefined.

This goes to my base question if you have a valid x,y,z coordinate we ought to have a valid IR value. In fact I would expect that we ought to always have a valid IR value.

xlz, 2017-01-14:
If both values into atan2 are close to zero this pixel is supposed to be invalid anyway.

> if you have a valid x,y,z coordinate we ought to have a valid IR value

Why? Sensor has limitations, you know?

mpatalberta, 2017-01-15:
Lingzhu
the Nvidia atan2 is not defined for the case of both a and b being small.

I would suggest if both values are small the resulting value for atan2 be forced to 1 and atan2 must not be called.

I am looking into a reason for intensity being zero with a valid x,y,z value.

Thanks

Mobile Patrick Brown
631 880 1188
1 Locust AVE North
Medford NY
11763 USA

- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

mpatalberta, 2017-01-16:
Took a little time but I now I see what is happeneing. I notice that I have dead pixels in my camera and thus dead intensity value. One camera I had 840 another camera 130. The location on both cases is that the pixel was at the side of the camera.  The kinectV2_bridge by the use of the interpolator on both the xyz and intensity value. An option needs to be added to change the behavior.

closed 2017-01-16 by mpatalberta