OpenKinect/libfreenect2 #782: Question: No ready-made solutions/plugins for UE4?

kusokuso1, 2017-01-19:
This isn't a bug report.

I've been trying to follow the instructions here https://github.com/OpenKinect/libfreenect2/blob/master/README.md in an effort to get the **Kinect v2 for Windows** to do something, anything, in Unreal Engine 4 on a Windows 7 64-bit machine.

Though some time has passed, it I can't seem to find any prebuilt/ready-made solutions/plugins, so it looks like I have to build a solution myself.

Problem is, I'm not familiar with building solutions.

Following the instructions, from top to bottom, I get stuck here:

**Build libusb**
Open a Git shell (GitHub for Windows), or any shell that has access to git.exe and msbuild.exe

cd depends/
.\install_libusb_vs2013.cmd

Typing the above (cd depends/ or .\install_libusb_vs2013.cmd or both)into Git Shell I get an error that folder "depends" doesn't exist.  That makes sense, considering I've never specified software be installed in a depends directory, nor to my recollection has anything ever been installed into such a directory.

I have both VS 2013 and 2015 community installed.

I apologize for my newbness but if anyone could help me it'd be extremely appreciated!

xlz, 2017-01-20:
> folder "depends" doesn't exist

Did you download the source code?

kusokuso1, 2017-01-31:
Hello xlz,
Sorry for my late reply and for asking such a stupid question.
I got my repositories confused.  Having downloaded the proper one, containing the depends folder, it all went smoothly.

Thank you for your time.  I hope I didn't waste too much of it!

RajaMOeez, 2017-01-31:
Hi Kusokuso Im also facing the same problem if any help from you side then it will be highly appreciated from my side. 
Best Regards 
MOeez Raja

RajaMOeez, 2017-01-31:
Hello Mr LingZhu I don't know how to build libusb using git shell i have tried but nothing I can dig up if any help that I'll be very great-full to you.

closed 2017-02-20 by xlz