OpenKinect/libfreenect2 #783: Protonect packet loss without rgb stream

gaborpapp, 2017-01-20:
Overview Description:

Packet loss with no rgb stream, while everything works fine when all streams are enabled.

Macbook pro, NVidia GT 750M, OS X 10.12, latest git, cec93c7

Steps to Reproduce:

1. `./Protonect -gpu= cuda -norgb`

Actual Results:

Many packets are lost, the depth stream is displayed at about 0.5-1 Hz.

[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] 71 packets were lost
[Info] [DepthPacketStreamParser] 64 packets were lost
[Info] [DepthPacketStreamParser] 242 packets were lost
[Info] [DepthPacketStreamParser] 74 packets were lost
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling depth transfers...

When the rgb stream is present I also get some lost packets, but the stream is smooth.

[Info] [DepthPacketStreamParser] 4 packets were lost
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [DepthPacketStreamParser] 2 packets were lost

The same thing happens with cl, cuda, cudakde depth processors.
Does `SyncMultiFrameListener` work without an rgb stream? Should I wait for the frames in a non-synchronized manner instead?
Any help would be appreciated.

gaborpapp, 2017-01-20:
I'm seeing the same behavior when using my own, non-synchronized `FrameListener` class. Without the rgb stream everything slows down substantially. Any pointers would be appreciated.

xlz, 2017-01-22:
What's the purpose of adding this argument `-gpu=`?

gaborpapp, 2017-01-22:
For some reason `-gpu=cuda` does not work for me, only `-gpu= cuda`. Since I'm on a NVidia Mac, the default OpenGL pipeline crashes the machine (#205), so changing the pipeline is necessary.

xlz, 2017-01-22:
I don't think you need `-gpu=` at all. And this argument `-gpu=` by itself is wrong.

gaborpapp, 2017-01-22:
Without that `Protonect` crashes for me with the OpenGL pipeline. Using it like above works on my mac, changes the pipeline correctly. It might be a mac thing how the arguments are handled. I can investigate this further in another issue if required, but I don't think it is related to the packet loss issue.

xlz, 2017-01-23:
`-gpu=` only specifies the id of the GPU to use for any GPU pipelines and it must be specified in the format of `-gpu=0` or `-gpu=1`. A singular `-gpu=` is undefined behavior.

xlz, 2017-01-23:
By the way, you could have provided just a little bit fuller log because the GPU information in it was truncated by you.

gaborpapp, 2017-01-23:
Thanks for the explanation about `-gpu=`. My bad. I merged the two consecutive parameters in the help.

The log files for cuda, opencl with and without rgb stream are here: https://gist.github.com/gaborpapp/c8fe4844699422e44253057a61c076de

xlz, 2017-01-23:
Use `-gpu=1` or `-gpu=2`.

gaborpapp, 2017-01-23:
The result is the same.

gaborpapp, 2017-01-31:
Tried this with an Intel HD 6000 gpu with linux and the OpenGL depth processor and could not reproduce the issue.

closed 2017-02-20 by xlz

gaborpapp, 2017-02-20:
I'm still seeing this issue on a mac. I'm sorry it got closed without a solution.

xlz, 2017-02-20:
There isn't really anything to do anymore towards a complete resolution. Macs with Nvidia graphics are always a hit and miss. It's almost always some issues in the MacOS kernel and without the source code there is little to do.