OpenKinect/libfreenect2 #785: libfreenect2 test error

QinZiwen, 2017-01-22:
I use ubuntu 14.04 and cuda 7.5 has been install.But when I run "./bin/Protonect ", terminal show log as follow:
```
$ ./bin/Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 021468743547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 1024
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 021468743547
device firmware: 4.0.3911.0
[Info] [TurboJpegRgbPacketProcessor] avg. time: 23.3263ms -> ~42.87Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 23.7407ms -> ~42.1217Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.9693ms -> ~40.0491Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.4328ms -> ~40.9285Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.5604ms -> ~40.716Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.8512ms -> ~40.2394Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.8406ms -> ~40.2566Hz
```
But the window of showing image disintegrate and my system tell me that dependence error. Picture of bottom show details.

xlz, 2017-01-23:
Run with `-noviewer` first. It looks like your USB ha some issues. https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#linux

QinZiwen, 2017-02-08:
My computer has a "i7 cpu and gtx1070", which is a notebook.

```
$ turtlebot@turtlebot-OMEN-by-HP-Laptop:~/ThridPKG/libfreenect2/build/bin$ ./Protonect -noviewer
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:7 with serial 021468743547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 021468743547
device firmware: 4.0.3911.0
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.5076ms -> ~40.8036Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.606ms -> ~40.6405Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.0847ms -> ~41.5202Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.9886ms -> ~40.0183Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 22.6887ms -> ~44.0748Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.6314ms -> ~40.5986Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.8978ms -> ~40.1643Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 25.0253ms -> ~39.9595Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.9015ms -> ~40.1581Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 22.0213ms -> ~45.4106Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.6534ms -> ~40.5624Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.9554ms -> ~40.0715Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.1684ms -> ~41.3764Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 25.2204ms -> ~39.6505Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 22.5561ms -> ~44.3339Hz
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.7048ms -> ~40.478Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 25.2653ms -> ~39.58Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.7672ms -> ~40.376Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 25.1421ms -> ~39.7739Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.984ms -> ~40.0256Hz
```

Only connect kinect2 and run "lsusb -t"
```
turtlebot@turtlebot-OMEN-by-HP-Laptop:~$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
    |__ Port 2: Dev 8, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 9, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 9, If 1 , Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 9, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 9, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 2: Dev 11, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 4: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 4: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 7: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M
```
Is this I connect ok?

```
$ turtlebot@turtlebot-OMEN-by-HP-Laptop:~$ lspci | grep USB
00:14.0 USB controller: Intel Corporation Device a12f (rev 31)
```

xlz, 2017-02-20:
> Intel Corporation Device a12f
> Intel Corporation Sunrise Point-H USB 3.0 xHCI Controller [8086:a12f]

I think your laptop uses skylake chipsets but the kernel drivers are not new enough to support the USB 3 controller. Try newer kernel http://packages.ubuntu.com/zesty/linux-generic

QinZiwen, 2017-02-21:
Yes,kinect can work and show image. Thanks for your help!

closed 2017-02-21 by xlz