OpenKinect/libfreenect2 #786: Uninstall libfreenect2 and reinstall with -DENABLE_CXX11=ON flag

pallavbakshi, 2017-01-23:
Hi,

I am using
1. Ubuntu 14.04
2. ROS Indigo

I installed libfreenect2 and forgot to use the flag `-DENABLE_CXX11=ON` flag during the cmake step. I tried looking for steps to uninstall libfreenect2 but there is nothing mentioned in the documentation. How can I uninstall it? And if there is any better way to have the ENABLE_CXX11=ON without uninstalling then that will be even better. 

Thanks

pallavbakshi, 2017-01-24:
and by the way, why have you given the option to install libopenni2-dev. On the webpage of openni2_camera it is mentioned that it won't be useful with Kinect v2 cameras.

xlz, 2017-01-25:
libfreenect2 is installed to a single directory. Delete the directory to uninstall.

This project is not related to openni2_camera so that webpage is irrelevant.

closed 2017-01-25 by xlz