OpenKinect/libfreenect2 #788: Kinect 1520 on Mac Sierra - No camera device

abeythomas, 2017-01-25:
[image]

No idea about this error.  Was just following the steps.
And by the way I have USB.3.0 and the depth is working while testing in Processing.
Please help!

xlz, 2017-01-25:
The steps don't cover the case where you already have a libfreenect2 directory in place.
`rm -r ~/libfreenect2` and try again.

closed 2017-01-25 by xlz

abeythomas, 2017-01-25:
Kinect 1520 on mac - No Camera device(continued) #4

[image]

This is what I got after removing the libfreenect2 folder and installing again
(Sorry for the trouble, I'm new to all these troubleshooting)
Please let me know know if there's a solution. Thanks

xlz, 2017-01-25:
Did you follow this step?

> Make sure these build tools are available: wget, git, cmake, pkg-config. Xcode may provide some of them. Install the rest via package managers.

abeythomas, 2017-01-25:
Yes! working now. Thanks a lot. 
I thought pkg-config was already there with Xcode, because it came up while searching in spotlight.