OpenKinect/libfreenect2 #792: add additional field to structure

mpatalberta, 2017-02-02:
In the depthpackaetprocess,h can I safely add a int64_t value.
(see below)
#ifndef DEPTH_PACKET_PROCESSOR_H_
#define DEPTH_PACKET_PROCESSOR_H_

#include <stddef.h>
#include <stdint.h>

#include <libfreenect2/config.h>
#include <libfreenect2/libfreenect2.hpp>
#include <libfreenect2/frame_listener.hpp>
#include <libfreenect2/packet_processor.h>

namespace libfreenect2
{

/** Data packet with depth information. */
struct DepthPacket
{
  uint32_t sequence;
  uint32_t timestamp;
  unsigned char *buffer; ///< Depth data.
  size_t buffer_length;  ///< Size of depth data.

  Buffer *memory;

 int64_t myTimeNOW;/// realworld tick count <<<<< new field
};

Expected Results:

Reproducibility:

Additional Information:

floe, 2017-02-03:
I don't quite get your point. Do you want this change included into libfreenect2? If so, please send a pull request (but why would you want that field anyway?).

mpatalberta, 2017-02-03:
I want to add the field to hold a realworld time. This is to allow coordination between multiple kinects for pictures taken at the same as near as possible.

xlz, 2017-02-20:
Sure. It's internal API anyway. 

Though this gets me the headache of worrying about choosing either pure C++11 or portability kludge again.

xlz, 2017-04-04:
Accurate time tracking is actually non-trivial. Temporally everything happens in this order:

IR:

1. Camera takes 3 frames, fixed interval, documented somewhere(?)
2. Onboard processing, relatively fixed amount of time, unknown
3. USB iso transfer, fixed amount of time, can be computed from USB spec
4. Decoding, varying amount of time

RGB:

1. Camera takes a frame with rolling shutter
2. Onboard processing/JPEG encoding, relatively fixed amount of time, unknown
3. USB bulk transfer, varying amount of time
4. Decoding, varying amount of time

We don't actually know where the device timestamps are taken. But in theory the best place to take it would be the moment the frame is collected.

The most reliable place to take realtime timestamp is when the first byte (or the first unit of data) of a USB transfer is complete.

If RGB and IR share the same clock onboard we can use the precise latency of USB iso transfer to derive the synchronization between device timestamp and walltime.

Accurate timing functions though, I still have doubt about C++11's high_resolution_clock on Windows as Windows' timing API was notoriously bad.

mpatalberta, 2017-04-05:
I have changed the libfreenect code to put a time stamp upon arrival of a packet from the usb driver in libfreenect.

- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

timprepscius, 2017-06-07:
@mpatalberta 

Could you post a diff of the code to place this timestamp in?  I am also interested in this code change.

mpatalberta, 2017-06-08:
I can send you the complete code if you want.
Who do I send it to.

To change I made was to get the time up reception of the packet of the usb data.
Changes are required in both the openfreenect2 and kinectv2.
The code of kinectv2 allows turning the camera off under neath with out loosing the kinectv2 session. I have that change also.

Pat,

timprepscius, 2017-06-08:
would you be sending a diff of a repository?  Or would you be sending actual files?

If a diff, and not too large, I wonder if gist or pastebin would be appropriate? (and then put a link here) . Or just attach to one of these github comment email/messages?

If actual files, I can make something available, and I'll do the diff etc.