OpenKinect/libfreenect2 #793: Could not find PkgConfig, JPEG, TegraJPEG and TurboJPEG (Windows 10, VS2015)

lin321inori, 2017-02-08:
Hi,
I have followed the instructions for Windows but encountered the following cmake problems. 

-- using tinythread as threading library
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE)
CMake Warning at cmake_modules/FindLibUSB.cmake:55 (MESSAGE):
UsbDk device driver is not found.   Fall back to libusbK.
Call Stack (most recent call first):
CMakeLists.txt:93 (FIND_PACKAGE)

-- Could NOT find JPEG (missing:   JPEG_LIBRARY  JPEG_INCLUDE_DIR)
-- Could NOT find TegraJPEG (missing:   TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS)
CMake Error at C:/Users/linin/Downloads/cmake-3.8.0-rc1-win64-x64/share/smake-3.8/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
Could NOT find TurboJPEG (missing: TurboJPEG_INCLUDE_DIRS TURBOJPEG_WORKS)
Call Stack (most recent call first):
C:/Users/linin/Downloads/cmake-3.8.0-rc1-win64-x64/share/smake-3.8/Modules/FindPackageHandleStandardArgs.cmake:337 (_FPHSA_FAILURE_MESSAGE):
cmake_modules/FindTurboJPEG.cmake:66 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
CMakeLists.txt:240 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!

Thanks in advance.

RajaMOeez, 2017-02-14:
hello Im also facing the same problem

RajaMOeez, 2017-02-18:
Im also facing the same error Sir. If you have find the solution please let me know, I'll be grateful to you. If you didn't get solution. In future if I got, I'll let you know.

xlz, 2017-02-20:
```
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: C:/libjpeg-turbo64/lib/turbojpeg.lib
[...]
 C:/libjpeg-turbo64/lib/turbojpeg.lib
```
Can't reproduce.

singularTnT, 2017-03-05:
Maybe you download the wrong version of TurboJPEG,
at this site,  [https://sourceforge.net/projects/libjpeg-turbo/files/1.5.1/](url)
This one [libjpeg-turbo-1.5.1-vc64.exe] is effective in x64 windows，
Hope you get it!

xlz, 2017-03-28:
Can't reproduce.

closed 2017-03-28 by xlz

sad490, 2017-08-17:
https://sourceforge.net/projects/libjpeg-turbo/files/1.5.1/

jennadeboisblanc, 2019-04-14:
I'm also running windows 10, 64-bit. Have installed libjpeg-turbo 1.5.1. Same error message.

Fatemeh-MA, 2019-08-08:
Hi, I had the same problem. you need to use "libjpeg-turbo-vc" version instead of "libjpeg-turbo-gcc".  it works for me. 
best luck